chipsalliance/rocket-chip #1360: How do openocd,gdb,spike work tegether?

clx7913, 2018-04-19:
My gdb version is
> GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.3) 7.7.1

`spike pk hello` works well.

Following https://github.com/freechipsproject/rocket-chip/pull/213 and https://riscv.org/software-tools/risc-v-isa-simulator/,I installed the openocd (version 0.10.0) by typing:
> git clone git://git.code.sf.net/p/openocd/code openocd
> sudo ./bootstrap
> sudo ./configure  --enable-jtag_vpi --enable-remote-bitbang
> make
> sudo make install

After I followed these lines(from https://riscv.org/software-tools/risc-v-isa-simulator/),I create spike.cfg in /openocd/tcl/target and typed  these lines in it:
> interface remote_bitbang
> remote_bitbang_host localhost
> remote_bitbang_port 9824
> 
> set _CHIPNAME riscv
> jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10e31913
> 
> set _TARGETNAME $_CHIPNAME.cpu
> target create $_TARGETNAME riscv -chain-position $_TARGETNAME
> 
> gdb_report_data_abort enable
> 
> init
> halt

But when I typed
> $ openocd -f spike.cfg
The terminal showed:

> Open On-Chip Debugger 0.10.0+dev-00403-g3737dd6 (2018-04-19-23:06)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
> Info : only one transport option; autoselect 'jtag'
> embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx, quark_d20xx, stm8,  or aarch64
> in procedure 'script' 
> at file "embedded:startup.tcl", line 60
> in procedure 'target' called at file "spike.cfg", line 9
> in procedure 'ocd_bouncer' 
> at file "embedded:startup.tcl", line 21
> 

I don't how do openocd,gdb,spike work tegether and what is the benefit.Could someone help me?

mwachs5, 2018-04-19:
It looks like your `openocd` command is picking up the OpenOCD that was already installed on your machine, not the one you built above. Make sure you are using the one that you actually installed with the above commands, by default it will be installed in the same directory that you downloaded it to, look for the `bin` directory there, then do `path-to-install-location/bin/openocd` instead of just `openocd`.

clx7913, 2018-04-20:
I looked for the openocd.There was just one location `/usr/local/bin/openocd`,Did you mean I should use `/usr/local/bin/openocd` instead of `openocd`?

But this kind of approch didn't work.

root@c:/home/chang/Downloads/openocd# which openocd
`/usr/local/bin/openocd`
root@c:/home/chang/Downloads/openocd# /usr/local/bin/openocd -f spike.cfg 
```
Open On-Chip Debugger 0.10.0+dev-00403-g3737dd6 (2018-04-19-23:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx, quark_d20xx, stm8,  or aarch64
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "spike.cfg", line 9
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21
```

Did I make any mistakes?

clx7913, 2018-05-27:
I changed the version of rocket-chip,and now the problem has gone.

closed 2018-05-27 by clx7913