chipsalliance/rocket-chip #1377: Error while run "make -j4 run" in /vsim

x-tit, 2018-04-25:
Hello,
I have access to VCS but while I run "make -j4 run " in /vsim ,I encounter the following error.
It seems that something is wrong with VCS library.What should I do?
Thanks a lot.

... ...
All of 168 modules done
make[1]: warning: -jN forced in submake: disabling jobserver mode.
make[1]: Entering directory '/mnt/house/rocket-chip/vsim/csrc'
make[1]: Leaving directory '/mnt/house/rocket-chip/vsim/csrc'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/mnt/house/rocket-chip/vsim/csrc'
g++ -w  -pipe -fPIC -I/mnt/house/VCS/vcs_2016.06//include -I/mnt/house/riscv-toolchain//include -std=c++11 -Wl,-rpath,/mnt/house/riscv-toolchain//lib -O -I/mnt/house/VCS/vcs_2016.06/include    -c /mnt/house/rocket-chip/src/main/resources/csrc//SimDTM.cc
g++ -w  -pipe -fPIC -I/mnt/house/VCS/vcs_2016.06//include -I/mnt/house/riscv-toolchain//include -std=c++11 -Wl,-rpath,/mnt/house/riscv-toolchain//lib -O -I/mnt/house/VCS/vcs_2016.06/include    -c /mnt/house/rocket-chip/src/main/resources/csrc//SimJTAG.cc
g++ -w  -pipe -fPIC -I/mnt/house/VCS/vcs_2016.06//include -I/mnt/house/riscv-toolchain//include -std=c++11 -Wl,-rpath,/mnt/house/riscv-toolchain//lib -O -I/mnt/house/VCS/vcs_2016.06/include    -c /mnt/house/rocket-chip/src/main/resources/csrc//remote_bitbang.cc
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C
/usr/bin/ld: /mnt/house/VCS/vcs_2016.06/linux64/lib/vcs_save_restore_new.o: relocation R_X86_64_32S against undefined symbol `_sigintr' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
Makefile:104: recipe for target 'product_timestamp' failed
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory '/mnt/house/rocket-chip/vsim/csrc'
Make exited with status 2
Makefrag:66: recipe for target 'simv-freechips.rocketchip.system-DefaultConfig' failed
make: *** [simv-freechips.rocketchip.system-DefaultConfig] Error 2

ferdous313, 2018-10-01:
I got the same error. I was wondering if you would please help me solve it.Thanks!

ferdous313, 2018-10-01:
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C
/usr/bin/ld: /mada/software/synopsys/vcs-mx/K-2015.09-SP2-14/linux64/lib/vcs_save_restore_new.o: relocation R_X86_64_32S against undefined symbol `_sigintr' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: nonrepresentable section on output
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:107: product_timestamp] Error 1
make[1]: Leaving directory '/mada/users/nferdous/boom-template/rocket-chip/vsim/csrc'
Make exited with status 2
make: *** [Makefrag:68: simv-freechips.rocketchip.system-DefaultConfig] Error 2

SunYongshuai, 2018-11-02:
This is the problem of vcs & gcc/g++.
When I change gcc&g++ to version 4.8, the error gone.
1. Install gcc-4.8 & g++4.8
2. Add '-cpp g++-4.8 -cc gcc-4.8' to VCS in Makefrag
3. May be ' -LDFLAGS -Wl,-no-as-needed ' is also needed.

closed 2020-02-09 by mwachs5