chipsalliance/rocket-chip #1379: Pull a signal to top level

ppoocde, 2018-04-25:
Hi everyone,

I want to pull a signal from TestHarness.dut.mbus level to TestHarness level so that I can write some logic statement to make use of this signal in TestHarness level?
For example, there is internal signal in mbus, let's say "signal_a", I want to create an output signal to connect to this signal_a, and this output will connect to another new output of dut. so TestHarness can have logic to make use of this output. 

Thanks,

edcote, 2018-04-26:
This is something that you could do with a FIRRTL [wiring transform](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/wiring/WiringTransform.scala).  FWIW, the question is not related to the generator.  Maybe try [chisel-users](https://groups.google.com/forum/#!forum/chisel-users)?

ppoocde, 2018-04-27:
Thanks, I will post other group

closed 2018-04-27 by ppoocde