chipsalliance/rocket-chip #1384: Error while run "make run CONFIG=BOOMConfig" in ../emulator

Royen102, 2018-04-28:
/home/RISC/rocket-chip/csrc/SimDTM.cc: In function ‘int debug_tick(unsigned char*, unsigned char, int*, int*, long long int*, unsigned char, unsigned char*, int, long long int)’:
/home/david/rocket-chip/csrc/SimDTM.cc:47:64: error: no matching function for call to ‘dtm_t::dtm_t(std::vector<std::__cxx11::basic_string<char> >)’
       dtm = new dtm_t(filter_argv_for_dtm(info.argc, info.argv));
                                                                ^
In file included from /home/david/rocket-chip/csrc/SimDTM.cc:3:0:
/home/RISC/riscv-tools/include/fesvr/dtm.h:17:3: note: candidate: dtm_t::dtm_t(int, char**)
   dtm_t(int argc, char**argv);
   ^
/home/RISC/riscv-tools/include/fesvr/dtm.h:17:3: note:   candidate expects 2 arguments, 1 provided
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-parentheses-equality’
VTestHarness.mk:69: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/RISC/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig'
Makefrag-verilator:68: recipe for target 'emulator-rocketchip-BOOMConfig' failed
make: *** [emulator-rocketchip-BOOMConfig] Error 2

mayoru, 2018-04-28:
[dtm interface](https://github.com/riscv/riscv-fesvr/commit/d8956c8f00d2dbba83e18f0f3a1c155031f158e5#comments) was changed significantly  a month ago, you should update your SimDTM.cc or downgrade your riscv-fesvr

Royen102, 2018-04-28:
@mayoru  Thank U for your answer，but my riscv-fesvr already up-to-date。

Royen102, 2018-04-28:
@ccelio  thank u，my riscv-tools was installed to a folder individually，rather than within rocket-chip/riscv-tools。Its details：
david@ubuntu:~/riscv-tools$ riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/david/riscv-tools/libexec/gcc/riscv64-unknown-linux-gnu/7.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/david/riscv-tools/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/home/david/riscv-tools --with-sysroot=/home/david/riscv-tools/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-nls --disable-bootstrap --src=.././riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc
Thread model: posix
gcc version 7.2.0 (GCC) 
david@ubuntu:~/riscv-tools$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/david/riscv-tools/libexec/gcc/riscv64-unknown-elf/7.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/david/riscv-tools/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/david/riscv-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/david/riscv-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=../../riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 7.2.0 (GCC)

Royen102, 2018-04-28:
david@ubuntu:~/riscv-tools/riscv-fesvr$ git branch
* （头指针分离于 68c12d0）
  master
david@ubuntu:~/rocket-chip$ git branch
* master

mayoru, 2018-04-28:
what about the commit versions
git rev-parse HEAD

Royen102, 2018-04-28:
@mayoru  
david@ubuntu:~/rocket-chip$ git rev-parse HEAD
d4ff29f0fd6e23fed46dd2fe7d7dd08b3c31f295
david@ubuntu:~/riscv-tools/riscv-fesvr$ git rev-parse HEAD
68c12d06ebbdfe20856b886570822fe66804fd26

Royen102, 2018-04-28:
@mayoru error while run : sbt compile
david@royen:~/rocket-chip/chisel3$ sbt compile
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/david/rocket-chip/chisel3/project
[info] Loading settings from build.sbt ...
[info] Set current project to chisel3 (in build file:/home/david/rocket-chip/chisel3/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Updating coreMacros...
[warn] 	module not found: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/david/.ivy2/local/edu.berkeley.cs/firrtl_2.11/1.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/firrtl_2.11/1.2-SNAPSHOT/firrtl_2.11-1.2-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/david/.sbt/preloaded/edu.berkeley.cs/firrtl_2.11/1.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/david/.sbt/preloaded/edu/berkeley/cs/firrtl_2.11/1.2-SNAPSHOT/firrtl_2.11-1.2-SNAPSHOT.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:firrtl_2.11:1.2-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:coremacros_2.11:3.2-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:10)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:28)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:242)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:190)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:46)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:99)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:112)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:112)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:95)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:126)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2383)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (coreMacros / update) sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] Total time: 5 s, completed 2018-4-28 19:41:35

mayoru, 2018-04-28:
Your commits seem to be up-to-date, but you also have _/rocket-chip/csrc_ folder, which has been moved to _src/main/resources_ recently. Check why files in your rocket-chip folder are different from the latest version d4ff29f.

If you have _rocket-chip/src/main/resources/csrc/folder_ and files in it, try to change emulator/Makefile as suggested [here](https://github.com/freechipsproject/rocket-chip/pull/1376/files) and remove _/rocket-chip/csrc_ folder, maybe it will help.

Royen102, 2018-04-28:
@mayoru  Thank U very much！ I‘ll try it.

ccelio, 2018-04-28:
Since you are using the latest rocket-chip commit, you should be able to follow the README instructions to build and test the `DefaultConfig` and prove that your riscv-fesvr and toolchain works with your rocket-chip setup. 

If you want to use BOOMConfig, that is not available in rocket-chip master. You can visit (https://github.com/ucb-bar/riscv-boom/releases) for the set of "stable" releases, which will point you to this README for v2.0.1:

https://github.com/ucb-bar/riscv-boom/tree/e71168e1b6412a1df85eedcabe2eecbae40bcf11

If you want to use the latest BOOM (which is very unstable) you can instead start here: https://github.com/ccelio/boom-template

In both cases, you need to build the riscv-tools that is found within `rocket-chip/riscv-tools`.

closed 2018-04-30 by Royen102