chipsalliance/rocket-chip #1385: error while run : sbt compile

Royen102, 2018-04-28:
david@ubuntu:~/rocket-chip/chisel3$ sbt compile
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/david/rocket-chip/chisel3/project
[info] Loading settings from build.sbt ...
[info] Set current project to chisel3 (in build file:/home/david/rocket-chip/chisel3/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Updating coreMacros...
[warn] 	module not found: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/david/.ivy2/local/edu.berkeley.cs/firrtl_2.11/1.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/firrtl_2.11/1.2-SNAPSHOT/firrtl_2.11-1.2-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/david/.sbt/preloaded/edu.berkeley.cs/firrtl_2.11/1.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/david/.sbt/preloaded/edu/berkeley/cs/firrtl_2.11/1.2-SNAPSHOT/firrtl_2.11-1.2-SNAPSHOT.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:firrtl_2.11:1.2-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:coremacros_2.11:3.2-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:10)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:28)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:242)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:190)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:46)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:99)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:112)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:112)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:95)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:126)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2383)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (coreMacros / update) sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] Total time: 5 s, completed 2018-4-28 19:41:35

mayoru, 2018-04-28:
Have you run 
sbt publish-local
in rocket-chip/firrtl directory?

clx7913, 2018-04-29:
I encountered the same problem with Royen102.

When I run `sbt publish-local` in rocket-chip/firrtl,it shows me:

```
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/chang/fpga-zynq/rocket-chip/firrtl/project
[info] Loading settings from build.sbt ...
[info] Set current project to firrtl (in build file:/home/chang/fpga-zynq/rocket-chip/firrtl/)
[error] Not a valid command: publish-local
[error] Not a valid project ID: publish-local
[error] Expected ':'
[error] Not a valid key: publish-local (similar: publishLocal, publishTo, publish)
[error] publish-local
[error]              ^
```

jackkoenig, 2018-04-29:
Usually rocket-chip is built using the Makefiles in the `vsim` and `emulator` directories. This flow will build `firrtl.jar` and put it in the `rocket-chip/lib` directory where it chisel3 can see it during compilation.

However, if you wish to build `chisel3` in isolation, you can publish firrtl as @mayoru said, the only issue is that `publish-local` is the old command. Now that we're using SBT > 1.0, you should use `publishLocal`.

clx7913, 2018-04-30:
@jackkoenig  So it is! Thank you for your guidance!

Royen102, 2018-04-30:
@mayoru @jackkoenig  Thanks! It's great!

closed 2018-04-30 by Royen102