chipsalliance/rocket-chip #1392: Some doubts about interrupt.

clx7913, 2018-04-30:
I want to implement timer interrupt.

After I use `asm volatile("li t0,0x08\n" "csrrs zero,mstatus,t0\n")`,I successfully create the executable file.

But when I run it with `spike pk` ,the terminal tells me:
```
z  0000000000000000 ra 00000000000100e2 sp 000000007f7e9b40 gp 00000000000123d8
tp 0000000000000000 t0 0000000000000008 t1 0000000000000000 t2 0000000000000000
s0 000000007f7e9b50 s1 0000000000000000 a0 0000000000000001 a1 000000007f7e9b58
a2 0000000000000000 a3 0000000000000001 a4 0000000000000010 a5 0000000000000002
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 00000000000101a4 va 000000003002a073 insn       3002a073 sr 8000000200046020
An illegal instruction was executed!
```
I don't know what's wrong.Maybe I haven't get the Machine-mode priority?But how could I get the Machine-mode priority?By Trap?

Any suggesstion would be greatly appreciated.

aswaterman, 2018-04-30:
The problem is that mstatus is an M-mode CSR, but programs running on pk run in U-mode, so they do not have access to M-mode CSRs.

clx7913, 2018-05-01:
OK,I see.

Is there no way to possess M-mode by `pk` ?

If I want to possess M-mode,what need I do?

aswaterman, 2018-05-01:
The easiest thing to do is to not use pk.  You can write bare-metal programs that are linked to start at address ```0x80000000``` and they will execute in M-mode.  The programs in riscv-tests are all examples of this.

clx7913, 2018-05-01:
You mean `./fesvr pk` on zedboard also couldn't get M-mode?

How should I run the bare-metal program on zedboard(I only have this board)?

aswaterman, 2018-05-01:
You can just run the program with fesvr in place of pk.  Try building one of the riscv-tests, like rv64ui-p-add, then run it with ```./fesvr rv64ui-p-add```

clx7913, 2018-05-01:
Very nice guidance!

Let me have a try. (＾▽＾)

clx7913, 2018-05-01:
Forgive my ignorance.I still have some questions.

In riscv-tests directory if I run
 ```
    $ autoconf
    $ ./configure --prefix=$RISCV/target
    $ make
    $ make install
```
it means I build all the isa/ and benchmarks/  files,doesn't it ?

If I want just build my single asm file called "mytest.s",what options should I add to riscv64-unknown-elf-gcc?

clx7913, 2018-05-03:
The another more simple test:
```
#include "riscv_test.h"
#include "test_macros.h"

RVTEST_RV64M
RVTEST_CODE_BEGIN

.equ CONSTANT, 0xcafebabe

        li a0, CONSTANT

RVTEST_CODE_END 

RVTEST_DATA_BEGIN

RVTEST_DATA_END
```
It also gets this kind of  error:
*** FAILED *** (tohost = 668)
```
C0:      33337 [1] pc=[0080000040] W[r30=0000000080001040][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00001f17] auipc   t5, 0x1
*** FAILED *** via dtm (code = 668, seed 1525289145) after 33348 cycles
root@c:/home/chang/fpga-zynq/rocket-chip/emulator# 
```

clx7913, 2018-05-03:
Resolved,that because the format of my program is not completely right.

Thanks again!

closed 2018-05-03 by clx7913

clx7913, 2018-05-09:
Sir,can I ask you why linking to start at address 0x80000000 will make the program execute in M-mode?

> The easiest thing to do is to not use pk. You can write bare-metal programs that are linked to start at address 0x80000000 and they will execute in M-mode. The programs in riscv-tests are all examples of this.