chipsalliance/rocket-chip #1397: Upstreaming utilities to chisel3

ducky64, 2018-05-01:
This has been talked about on and off for quite a while now, but some of the utilities in rocket-chip, like the async FIFOs, are quite general and would be good candidates for inclusion in a chisel standard library, which is being discussed in freechipsproject/chisel3#818

I'd just like to get thoughts from the rocket people about including these in a chisel standard library, which other parts could be useful to include, and how it could be done.

hcook, 2018-05-02:
In approximate order from you-really-ought-to down through might-be-nice-to-have:

Better muxing logic for literal constants:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/MuxLiteral.scala

Collections where every element is the same type but differently parameterized:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/HeterogeneousBag.scala

All these implicit classes should have their methods added to the original Chisel class (IMO):
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/package.scala

Async reset registers (but perhaps reimplemented to use parameterized black boxes):
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/AsyncResetReg.scala

PRNG generation in HW (@seldridge mentioned his own version in the chisel issue):
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/LCG.scala
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/Frequency.scala
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/Misc.scala

Various other helper objects from this miscellaneous file:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/Misc.scala

codelec, 2018-05-22:
an optimized 2-entry fluid queue https://github.com/masc-ucsc/fluid/blob/master/src/main/scala/elastic-reg.scala

edcote, 2018-05-24:
I vote for Diplomacy.

sequencer, 2022-04-08:
diplomacy is ongoing!
splitting rocket utils to chisel3 standard lib is certainly on its way.
we need a concrete plan on doing this.