chipsalliance/rocket-chip #1398: LazyRoCC broken on master: Classes defined in method bodies are not supported.

morse-michael, 2018-05-02:
When building rocket-chip with WithRoccExample I hit the following error post-elaboration: org.json4s.package$MappingException: Classes defined in method bodies are not supported.

Going back through history it looks like the following pull request is responsible: https://github.com/freechipsproject/rocket-chip/pull/1369

When building rocket with WithRoccExample on commit e38898e (pull request #1345) it works fine, when I build it on commit 95f07a8 (pull request #1358) it breaks, see commits below for reference:

```
| *   95f07a8 Merge pull request #1358 from freechipsproject/param-desc-anno
| |\  
| | * b5b7e85 (github/param-desc-anno) annotations: simplify imports
| | * e87965a annotations: add TopLevelPortAnnotation
| | * 1687a87 annotations: enhance AddressMapAnnotation
| | * ed8caa9 annotations: more comments
| | * d58a575 annotations: first cut at two different mapping annotations
| | * 127c5f8 annotations: add AddressMapAnnotation
| | * 55d559a annotations: add Params annotations to various modeuls
| | * e5e7de8 annotations: add ParamsAnnotation wrapper
| |/  
| *   e38898e Merge pull request #1345 from freechipsproject/optional-ports
```

Looks like this commit requires some changes in LazyRoCC.scala - any ideas?

Full error message below:
```
...
[[35minfo[0m] [20.851] Done elaborating.
[0m[[0m[31merror[0m] [0m[0m(run-main-0) org.json4s.package$MappingException: Classes defined in method bodies are not supported.[0m
[0m[[0m[31merror[0m] [0m[0morg.json4s.package$MappingException: Classes defined in method bodies are not supported.[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.package$.fail(package.scala:95)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder$$anonfun$createConstructorDescriptors$3$$anonfun$15.apply(Reflector.scala:145)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder$$anonfun$createConstructorDescriptors$3$$anonfun$15.apply(Reflector.scala:142)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$class.map(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.AbstractTraversable.map(Traversable.scala:104)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder$$anonfun$createConstructorDescriptors$3.apply(Reflector.scala:142)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder$$anonfun$createConstructorDescriptors$3.apply(Reflector.scala:136)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.mutable.ArraySeq.foreach(ArraySeq.scala:74)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.TraversableLike$class.map(TraversableLike.scala:234)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.collection.AbstractTraversable.map(Traversable.scala:104)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder.createConstructorDescriptors(Reflector.scala:136)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder.constructorsAndCompanion(Reflector.scala:121)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$ClassDescriptorBuilder.result(Reflector.scala:183)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$.createDescriptor(Reflector.scala:53)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$$anonfun$describe$1.apply(Reflector.scala:48)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$$anonfun$describe$1.apply(Reflector.scala:48)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.package$Memo.apply(package.scala:36)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.reflect.Reflector$.describe(Reflector.scala:48)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.decomposeObject$1(Extraction.scala:118)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:228)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.addField$1(Extraction.scala:110)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.decomposeObject$1(Extraction.scala:140)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:228)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:189)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.addField$1(Extraction.scala:110)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:164)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.addField$1(Extraction.scala:110)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.decomposeObject$1(Extraction.scala:140)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:228)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.internalDecomposeWithBuilder(Extraction.scala:189)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.Extraction$.decomposeWithBuilder(Extraction.scala:64)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.native.Serialization$.writePretty(Serialization.scala:54)[0m
[0m[[0m[31merror[0m] [0m[0m	at org.json4s.native.Serialization$.writePretty(Serialization.scala:49)[0m
[0m[[0m[31merror[0m] [0m[0m	at firrtl.annotations.JsonProtocol$$anonfun$serializeTry$1.apply(JsonProtocol.scala:51)[0m
[0m[[0m[31merror[0m] [0m[0m	at firrtl.annotations.JsonProtocol$$anonfun$serializeTry$1.apply(JsonProtocol.scala:51)[0m
[... 17 lines cut ...]
[0m[[0m[31merror[0m] [0m[0m	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)[0m
[0m[[0m[31merror[0m] [0m[0m	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)[0m
[0m[[0m[31merror[0m] [0m[0m	at java.lang.reflect.Method.invoke(Method.java:498)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run.invokeMain(Run.scala:93)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run.run0(Run.scala:87)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run.execute$1(Run.scala:65)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run.$anonfun$run$4(Run.scala:77)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.TrapExit$App.run(TrapExit.scala:252)[0m
[0m[[0m[31merror[0m] [0m[0m	at java.lang.Thread.run(Thread.java:748)[0m
[0m[[0m[31merror[0m] [0m[0mjava.lang.RuntimeException: Nonzero exit code: 1[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run$.executeTrapExit(Run.scala:124)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Run.run(Run.scala:77)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)[0m
[0m[[0m[31merror[0m] [0m[0m	at scala.util.Try$.apply(Try.scala:209)[0m
[0m[[0m[31merror[0m] [0m[0m	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)[0m
[0m[[0m[31merror[0m] [0m[0m	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)[0m
[0m[[0m[31merror[0m] [0m[0m	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)[0m
[0m[[0m[31merror[0m] [0m[0m	at java.lang.Thread.run(Thread.java:748)[0m
[0m[[0m[31merror[0m] [0m[0m(Compile / [31mrunMain[0m) Nonzero exit code: 1[0m
```

colinschmidt, 2018-05-02:
I also ran into this issue but do not have a solution for it. It feels like it may be a limitation of turning `Parameters` into annotations.
Since `Parameters` can be any scala object it seems like we will need to have some way to opt out of RocketTile making its parameters annotations or have the `Parameter->Annotation` check whether the Parameter it is serializing can be serialized (maybe start with only serializing case classes?).

morse-michael, 2018-05-02:
Thanks @colinschmidt, when I disable annotations for RocketTile the core is indeed building fine again for me with WithRoccExample enabled. I have done this by commenting out the following line in RocketTile.scala:

`  annotated.params(this, outer.rocketParams)
`

PR #1404 (2018-05-08): Update and simplify RoCC parameterization
I'm eliminating the somewhat convoluted RoccParams in favor of just directly supplying `BuildRoCC` generator functions and cleaning up the flow of parameters. This has the following consequences:

- `nPTWPorts` and `usesFPUs` are now parameters supplied by individual accelerators to the base `LazyRoCC` class
- each accelerator takes an `OpcodeSet` as a constructor argument, because it is now a mandatory argument to `LazyRoCC` (backwards incompatible, sorry!)
- `AccumulatorExample` actually takes `n` as an argument
- `BuildRoCC` is now completely indepe [... cut]
closed 2018-05-08 by hcook