chipsalliance/rocket-chip #1408: Is that possible to configure the I/Dcache to ITIM/DTIM in Rocket, how?

MeibaoXu, 2018-05-08:
Hi everyone:

    I'm totally new to RISCV, and I've beening looking into the rocket tile, as well as SiFive core IP specs. There are ITIM and DTIM in SiFive's cores, and ITIM looks configurable as is mentioned in the spec. Hopefully, I thought ITIM is also avaliable in the rocket core, even though its option is not included in current CONFIGs. 
    I spent some very hard time trying to understand the source code in chisel, and there was a 'itimAddr' member in the 'ICacheParams' class, which is located in the 'ICache.scala'. It is currently defined as a None Option, and I though when the 'itimAddr' is configured to a proper value, the ITIM would be generated. However, a lot of methods had be tried and not correct results. May there are mistakes in my scala code since I have no scala/Java base, or maybe it is not possible to get a ITIM in rocket now, because it is SiFive core only.
    There are also some variable named with 'dtim' in the 'DCache.scala', but I cannot understand these chisel file right now. Is it possible to get  ITIM/DTIM in rocket, or does anyone know something about theconfiguration, please help me.

BR 
Meibao

juliusbaxter, 2018-05-16:
As for the ITIM, to enable it all you need to do is add the itimAddr parameter to the cache and ensure the cache is multi-way. It must be multi-way because the ITIM functionality is available for all cache ways except the last. I was asking about this the other day and @sorear pointed me to section 4.11 of this document: https://static.dev.sifive.com/FU540-C000-v1.0.pdf which describes the ITIM functionality a little bit.
My understanding of the way it works is as follows: you enable the ITIM by allocating it an address in the system. This will add a slave port onto the ICache in the Frontend module. Say you have an 8-way cache, the bottom 7/8ths of the cache become writable. According to the document above you write your instructions in and as you do there are some flops which record the fact that those lines in the cache are allocated to ITIM (locked, effectively, although addressed only through the physical ITIM address). Once you've loaded your program in there you should be good to point the core's PC at the ITIM address space and execute. I've seen that work on Rocket so far.

It would be good to have someone in the know ack my understanding, though.

As for the DTIM, that's easy to enable and there are example configurations such as this: https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/Configs.scala#L331

Basically I think you specify an address called "scratch" in the DCacheParams and the nWays must be 1.

MeibaoXu, 2018-05-17:
Hi @juliusbaxter 

Thanks a lot, your reply is quite helpful.

I have beening studying  the SiFive E31 Core Complex these days, and found that the DTIM generation is simply enough to just fill the "scratch" with a valid address, as you has mentioned. And the new generated verilog will involve a new module named "dtim_adapter" connecting to Dcache.
And I had also found that the ITIM generation might also need an address assigned to the "itimAddr". 

However, no matter which address is allocated to "itimAddr", the make would always fail at memory assignment error, even though I followed the E31 spec to assign the 0x0800_0000 address to "itimAddr".
Now I have found what I have missed with your help, that is the "nWays", which is set to 1 in "With1TinyCore" and "WithNSmallCores" (I should try with "WithNBigCores" first to found this issue).

Now I set the "nWays" to 2, and the "itim" finially shows up  in the memory map:
Generated Address Map
`	       0 -     1000 ARWX  debug-controller@0
	    3000 -     4000 ARWX  error-device@3000
	   10000 -    20000  R XC rom@10000
	 2000000 -  2010000 ARW   clint@2000000
	 8000000 -  8002000 ARWX  itim@8000000
	 c000000 - 10000000 ARW   interrupt-controller@c000000
	60000000 - 80000000  RWX  mmio-port-axi4@60000000
	80000000 - 80004000 ARWX  dtim@80000000
	90000000 - a0000000  RWXC memory@90000000
`
And exactly as you has said, the "Frontend" is extended with a group of slave port connecting to icache:
[image]

BR
Meibao

closed 2018-05-17 by MeibaoXu