chipsalliance/rocket-chip #1409: Can L1 cache be removed for rocket-chip?

clx7913, 2018-05-09:


ppoocde, 2018-05-24:
u cannot remove L1 instruction cache. If your reason to remove cache is to reduce area, the smallest L1 I cache you can set without crashing the generator is nSets = 2, nWays = 1.
u cannot remove D cache but u can change L1 data cache to stretchpad memory.

clx7913, 2018-06-12:
Sorry for replying so late and thank you for your answer.

I am a rookie. May I ask you how to access(like "read" or "write") the cache lines by assembler program? Is this possible?

Any suggestion will be appreciated.

ppoocde, 2018-06-12:
If your d cache are stretchpad memory with base address 0x80000000
e.g. the TinyConfig for d cache
scratch = Some(0x80000000L)
You can access it using load store instruction.

clx7913, 2018-06-12:
Ok, I see.

If I do not replace the caches with stretchpad memory, how could I access those caches? Are there some pre-build assembler program macros specially used to do these things?