chipsalliance/rocket-chip #1410: How to modify the clock frequency of rocket-chip?

clx7913, 2018-05-09:
I've seen the .json file generated in emulator(using DefaultConfig):
`{"#address-cells":[1],"#size-cells":[1],"compatible":["freechips,rocketchip-unknown-dev"],"cpus":{"#address-cells":[1],"#size-cells":[0],"cpu@0":{"clock-frequency":[0],"compatible":["sifive,rocket0","riscv"],"d-cache-block-size":[64],"d-cache-sets":[64],"d-cache-size":[16384],"
...`
Why the "clock-frequency"  is 0.And how to modify the frequency of rocket-chip?

edcote, 2018-06-11:
This clock-frequency is not directly related to hardware implementation.  What you are seeing is the device tree file.  See https://elinux.org/Device_Tree_Reference for more info.  I don't believe that the rocket-chip project has any real notion of clock frequency.  You would take the verilog model generated by the tool and implement it using your platform of choice.

clx7913, 2018-06-12:
Thank you, that helps!

closed 2018-06-12 by clx7913