chipsalliance/rocket-chip #1417: make -jN run-asm-tests hanging

Generalbrus, 2018-05-11:
I've used `make debug` to compile rocket-chip, but on `make -j4 run-asm-tests` the tests don't succeed (they don't go through actually), it looks like they are left waiting for some kind of connection. I'm not quite sure if I should do something special or it's the pipeline's fault. Here the last lines of output I get, after which it's just idle:

```
./emulator-freechips.rocketchip.system-DefaultConfig +max-cycles=100000000 +verbose output/rv64um-v-mul 3>&1 1>&2 2>&3 | /usr/local/bin/spike-dasm  > output/rv64um-v-mul.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.system-DefaultConfig +max-cycles=100000000 +verbose output/rv64um-v-mulh 3>&1 1>&2 2>&3 | /usr/local/bin/spike-dasm  > output/rv64um-v-mulh.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.system-DefaultConfig +max-cycles=100000000 +verbose output/rv64um-v-mulhsu 3>&1 1>&2 2>&3 | /usr/local/bin/spike-dasm  > output/rv64um-v-mulhsu.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.system-DefaultConfig +max-cycles=100000000 +verbose output/rv64um-v-mulhu 3>&1 1>&2 2>&3 | /usr/local/bin/spike-dasm  > output/rv64um-v-mulhu.out && [ $PIPESTATUS -eq 0 ]
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 57443
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 57444
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 57445
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 57446
make: *** [output/rv64um-v-mulh.out] Error 1
make: *** Waiting for unfinished jobs....
make: *** [output/rv64um-v-mulhsu.out] Error 1
```

zGitHubj, 2019-12-07:
Hi!Have you solved this problem?
Thanks a lot!

larabr, 2019-12-10:
Nope, I gave up on this

closed 2020-02-09 by mwachs5