chipsalliance/rocket-chip #1419: Mixed core class

hoangt, 2018-05-11:
```
class WithNBigSmallCores(nbig: Int, nsmall: Int) extends Config((site, here, up) => {
  case RocketTilesKey => {
    val big = RocketTileParams(
      core   = RocketCoreParams(mulDiv = Some(MulDivParams(
        mulUnroll = 8,
        mulEarlyOut = true,
        divEarlyOut = true))),
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes))),
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        blockBytes = site(CacheBlockBytes)))
    )
    List.tabulate(nbig)(i => big.copy(hartId = i))

    val small = RocketTileParams(
      core = RocketCoreParams(useVM = false, fpu = None),		// no FPU in small core
      btb = None,
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBEntries = 4,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes))),
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBEntries = 4,
        blockBytes = site(CacheBlockBytes)))
		)
    List.tabulate(nsmall)(_ + nbig)(i => small.copy(hartId = i))
  }
})

```
I create WithNBigSmallCores to support both big and small cores in subsystem/Config.scala. I believe hardId for cores must be incremental. However, there is a syntax error for below line
List.tabulate(nsmall)(_ + nbig)(i => small.copy(hartId = i))

Any ideal to fix this ? Can we wrap "val small = RocketTileParams( ... ) in a for ... loop then direactly assign loop index to hardId.

/T

Tong-Lin, 2018-05-13:
Like this:
```
class WithNBigMSmallCores(n: Int, m:Int) extends Config((site, here, up) => {
   case RocketTilesKey => {
     val big = RocketTileParams(
       core   = RocketCoreParams(mulDiv = Some(MulDivParams(
         mulUnroll = 8,
         mulEarlyOut = true,
         divEarlyOut = true))),
       dcache = Some(DCacheParams(
         rowBits = site(SystemBusKey).beatBits,
         nMSHRs = 0,
         blockBytes = site(CacheBlockBytes))),
       icache = Some(ICacheParams(
         rowBits = site(SystemBusKey).beatBits,
         blockBytes = site(CacheBlockBytes))))
     val small = RocketTileParams(
       core = RocketCoreParams(useVM = false, fpu = None),
       btb = None,
       dcache = Some(DCacheParams(
         rowBits = site(SystemBusKey).beatBits,
         nSets = 64,
         nWays = 1,
         nTLBEntries = 4,
         nMSHRs = 0,
         blockBytes = site(CacheBlockBytes))),
       icache = Some(ICacheParams(
         rowBits = site(SystemBusKey).beatBits,
         nSets = 64,
         nWays = 1,
         nTLBEntries = 4,
         blockBytes = site(CacheBlockBytes))))
     List.tabulate(n)(i => big.copy(hartId = i))++
     List.tabulate(m)(i => small.copy(hartId = i+n))
   }
 })
```

hoangt, 2018-05-14:
@Tong-Lin: Many thanks, it works indeed.

closed 2018-05-14 by hoangt