chipsalliance/rocket-chip #1440: The purpose of the randomize inits

mayoru, 2018-05-21:
+define+RANDOMIZE_MEM_INIT
+define+RANDOMIZE_REG_INIT
+define+RANDOMIZE_GARBAGE_ASSIGN
+define+RANDOMIZE_INVALID_ASSIGN

Please comment on  #1227 issue. @edcote said

> those defines are to eliminate possible sources of "X's" and help root cause any issues with X propagation.

But shouldn't the RTL simulation work without those defines too and not give any X propagation? It doesn't work now. How can we be sure that the post-synthesis model works well then?
It is not an issue for an FPGA, where all the registers are reset by default, but what about an ASIC?

edcote, 2018-05-24:
The RTL simulation will inherently propagate X.  The why is outside of the scope of this project,  I google'ed a good introduction to the who/why/what/how of X's and how they are used in digital logic simulation: http://www.sutherland-hdl.com/papers/2013-DVCon_In-love-with-my-X_paper.pdf

How can you be sure that the post-synthesis model works?  Well, you hire a good verification engineer or purchase the right verification tool (e.g. https://www.mentor.com/products/fv/questa-x-check).

Btw, on an FPGA, memories are not necessarily initialized to zero.  Be careful making these assumptions.  I wouldn't necessarily assume that all FPGA registers are reset to zero.

mayoru, 2018-06-01:
Thanks, @edcote 
For the post-synthesis I can do logic equivalence checking. At least it gives some confidence. The problem is to be sure that the RTL model is correct.
So how is it guaranteed that the RTL model is correct if I run the simulation tests only for one random set of initial values?

edcote, 2018-06-03:
Can you please close this issue?  Your requestions are not relevant to rocket-chip repo.

I am not sure what guarantees you are looking for.  The verification strategy of replacing X's with $random initial values is done with the hope that a checker will better catch randomize values than X's or X's that become 0's through X optimism cases.

closed 2020-02-09 by mwachs5