chipsalliance/rocket-chip #1441: make file compilation failed

tassneemhelal, 2018-05-22:
I'm facing a problem, when I try to run the vsim make file or when I try to run the make verilog command and I wanted to know what is the problem and how can I fix it. 

These are the errors that I get:

tassneemhelal@tassneemhelal-OptiPlex-990:~/rocket-chip/vsim$ make 

mkdir -p /home/tassneemhelal/rocket-chip/vsim/generated-src/

cd /home/tassneemhelal/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/tassneemhelal/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/tassneemhelal/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig"

OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0

[info] Loading settings from plugins.sbt ...

[info] Loading project definition from /home/tassneemhelal/rocket-chip/project

[info] Loading settings from build.sbt ...

[info] Loading settings from build.sbt ...

[info] Loading settings from build.sbt ...

Using addons: 

[info] Set current project to rocketchip (in build file:/home/tassneemhelal/rocket-chip/)

[info] Compiling 30 Scala sources to /home/tassneemhelal/rocket-chip/chisel3/target/scala-2.11/classes ...

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:4:8: not found: object uncore

[error] import uncore._

[error]        ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:5:8: not found: object rocket

[error] import rocket._

[error]        ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:6:8: not found: object rocket

[error] import rocket.Util._

[error]        ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:9:29: not found: type ChiselConfig

[error] class DefaultConfig extends ChiselConfig (

[error]                             ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:156:26: not found: type ChiselConfig

[error] class With2Cores extends ChiselConfig(knobValues = { case "NTILES" => 2 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:157:26: not found: type ChiselConfig

[error] class With4Cores extends ChiselConfig(knobValues = { case "NTILES" => 4 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:158:26: not found: type ChiselConfig

[error] class With8Cores extends ChiselConfig(knobValues = { case "NTILES" => 8 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:160:26: not found: type ChiselConfig

[error] class With2Banks extends ChiselConfig(knobValues = { case "NBANKS" => 2 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:161:26: not found: type ChiselConfig

[error] class With4Banks extends ChiselConfig(knobValues = { case "NBANKS" => 4 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:162:26: not found: type ChiselConfig

[error] class With8Banks extends ChiselConfig(knobValues = { case "NBANKS" => 8 })

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:164:27: not found: type ChiselConfig

[error] class WithL2Cache extends ChiselConfig(

[error]                           ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:187:34: not found: type ChiselConfig

[error] class WithL2Capacity2048 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 2048 })

[error]                                  ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:188:34: not found: type ChiselConfig

[error] class WithL2Capacity1024 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 1024 })

[error]                                  ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:189:33: not found: type ChiselConfig

[error] class WithL2Capacity512 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 512 })

[error]                                 ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:190:33: not found: type ChiselConfig

[error] class WithL2Capacity256 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 256 })

[error]                                 ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:191:33: not found: type ChiselConfig

[error] class WithL2Capacity128 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 128 })

[error]                                 ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:192:32: not found: type ChiselConfig

[error] class WithL2Capacity64 extends ChiselConfig(knobValues = { case "L2_CAPACITY_IN_KB" => 64 })

[error]                                ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:194:31: not found: type ChiselConfig

[error] class DefaultL2Config extends ChiselConfig(new WithL2Cache ++ new DefaultConfig)

[error]                               ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:195:35: not found: type ChiselConfig

[error] class DefaultL2VLSIConfig extends ChiselConfig(new WithL2Cache ++ new DefaultVLSIConfig)

[error]                                   ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:196:34: not found: type ChiselConfig

[error] class DefaultL2CPPConfig extends ChiselConfig(new WithL2Cache ++ new DefaultCPPConfig)

[error]                                  ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:197:35: not found: type ChiselConfig

[error] class DefaultL2FPGAConfig extends ChiselConfig(new WithL2Capacity64 ++ new WithL2Cache ++ new DefaultFPGAConfig)

[error]                                   ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:199:26: not found: type ChiselConfig

[error] class FPGAConfig extends ChiselConfig (

[error]                          ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:205:33: not found: type ChiselConfig

[error] class DefaultFPGAConfig extends ChiselConfig(new FPGAConfig ++ new DefaultConfig)

[error]                                 ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:207:27: not found: type ChiselConfig

[error] class SmallConfig extends ChiselConfig (

[error]                           ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:229:35: not found: type ChiselConfig

[error] class ExampleMediumConfig extends ChiselConfig(new MediumConfig ++ new DefaultConfig)

[error]                                   ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:231:38: not found: type ChiselConfig

[error] class DefaultFPGASmallConfig extends ChiselConfig(new SmallConfig ++ new DefaultFPGAConfig)

[error]                                      ^

[error] /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala:233:34: not found: type ChiselConfig

[error] class ExampleSmallConfig extends ChiselConfig(new SmallConfig ++ new DefaultConfig)

[error]                                  ^

[error] 27 errors found

[error] (chisel / Compile / compileIncremental) Compilation failed

[error] Total time: 12 s, completed May 18, 2018 4:17:56 PM

/home/tassneemhelal/rocket-chip/vsim/Makefrag-verilog:12: recipe for target '/home/tassneemhelal/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.fir' failed

make: *** [/home/tassneemhelal/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.fir] Error 1

mayoru, 2018-05-22:
Try removing /home/tassneemhelal/rocket-chip/chisel3/PublicConfigs.scala. At least I don't have it in my Chisel3 directory. It seems to be an old configuration file for rocket.
Also check you have up-to-date Chisel3 and rocket-chip directories. Something wrong is with your repository.

tassneemhelal, 2018-05-23:
Thank you so much. It worked finally! @mayoru

closed 2018-05-23 by tassneemhelal