chipsalliance/rocket-chip #1445: How to generate a dual-core processor with a specific cache architecture?

tassneemhelal, 2018-05-23:
Hi
I just started using Rocket Chip generator and I'm a complete newbie to it and I don't know where should I begin to start configuring my implementation for a dual-core processor with a specific cache architecture. I already followed all the steps included in the README file. What should I do next?

Any help would be appreciated.

Thank You :)

ppoocde, 2018-05-24:
type "make CONFIG=DualCoreConfig all" at vsim can give u dual core processor

mayoru, 2018-05-24:
Amend these files for a desired configuration:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/Configs.scala
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/Configs.scala
Specifically, make your own classes based on DualCoreConfig class and WithNBigCores/WithNSmallCores classes with desired cache parameters

tassneemhelal, 2018-05-24:
@ppoocde 

I get an error when I try to run it 

cd . && \
rm -rf csrc && \
vcs -full64 -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet +rad +v2k +vcs+lic+wait +vc+list -CC "-I/include" -CC "-I/home/tassneemhelal/Documents/rocket-chip//include" -CC "-std=c++11" -CC "-Wl,-rpath,/home/tassneemhelal/Documents/rocket-chip//lib" /home/tassneemhelal/Documents/rocket-chip//lib/libfesvr.so -sverilog +incdir+/home/tassneemhelal/Documents/rocket-chip/vsim/generated-src +define+CLOCK_PERIOD=1.0 /home/tassneemhelal/Documents/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DualCoreConfig.v /home/tassneemhelal/Documents/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DualCoreConfig.behav_srams.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/TestDriver.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/SimDTM.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/SimJTAG.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/plusarg_reader.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/ClockDivider2.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/ClockDivider3.v /home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc/AsyncResetReg.v  /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimDTM.cc /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimJTAG.cc /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/remote_bitbang.cc +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +libext+.v  -o ./simv-freechips.rocketchip.system-DualCoreConfig \
-debug_pp \

vcs: '+lint=all,noVCDE,noONGS,noUI' is not a vcs command. See 'vcs help'.
Makefrag:66: recipe for target 'simv-freechips.rocketchip.system-DualCoreConfig' failed
make: *** [simv-freechips.rocketchip.system-DualCoreConfig] Error 1

tassneemhelal, 2018-05-24:
@mayoru I just have to edit in these files? or should I write my implementation in a new file?

mayoru, 2018-05-24:
Do you have Synopsys vcs installed? Have you successfully run "make run " (with DefaultConfig)?
The script sources all .scala files in your src/main/scala directory, so it's up to you where to keep your configuration classes.

tassneemhelal, 2018-05-24:
@mayoru No, I don't have Synopsys vcs installed and yeah I ran the make run successfully.

mayoru, 2018-05-24:
Then you should've built and run C emulator only:
cd emulator
make run

seldridge, 2018-05-25:
Broadly, you define a new System and (possibly) Subsystem configuration either by adding to the existing System/Subsystem configuration files or by defining your own configuration files that are capable of being found during the build process. There exist a number of canned configurations, like the `DualCoreConfig` referenced above.

[The build scripts look for an overridable Makefile variable, `CONFIG` that defines which System configuration to use](https://github.com/freechipsproject/rocket-chip/blob/master/Makefrag#L9). That is expected to be in the package `freechips.rocketchip.system` (but you can override the `PROJECT` variable if you intend to put it somewhere else). For ease, you can add your configurations inside `{system, subsystem}/Configs.scala`.

The configuration is done in a composable way. I'm purposefully glossing over details, but whatever configuration `Field` that you define first will win (like with pattern matching in Scala). This allows you to override any configuration`Field` by prepending (with `++`) a new subclass of `Config`. In effect, you start with an existing configuration and then tweak it to your needs. You see this used extensively in the System and Subsystem configurations. Take a look at how `DualCoreConfig` and `TinyConfig` are composed.

closed 2018-06-05 by tassneemhelal