chipsalliance/rocket-chip #1448: fatal error: no such file or directory

tassneemhelal, 2018-05-24:
Hi,

when I try to run the make command in the emulator directory I get this error, What could be the problem?

mkdir -p /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig
/home/tassneemhelal/Documents/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+RANDOMIZE_GARBAGE_ASSIGN +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 --output-split-cfuncs 20000 -Wno-STMTDLY --x-assign unique -I/home/tassneemhelal/Documents/rocket-chip/src/main/resources/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/tassneemhelal/Documents/rocket-chip//include -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/verilator.h -include /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.plusArgs" -Mdir /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig \
-o /home/tassneemhelal/Documents/rocket-chip/emulator/emulator-freechips.rocketchip.system-DefaultConfig /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.v /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v  /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/emulator.cc /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimDTM.cc /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimJTAG.cc /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/remote_bitbang.cc -LDFLAGS " -L/home/tassneemhelal/Documents/rocket-chip//lib -Wl,-rpath,/home/tassneemhelal/Documents/rocket-chip//lib -L/home/tassneemhelal/Documents/rocket-chip/emulator -lfesvr -lpthread" \
-CFLAGS "-I/home/tassneemhelal/Documents/rocket-chip/emulator/generated-src -include /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig/VTestHarness.h"
make VM_PARALLEL_BUILDS=1 -C /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory '/home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig'
g++  -I.  -MMD -I/home/tassneemhelal/Documents/rocket-chip/emulator/verilator/install/share/verilator/include -I/home/tassneemhelal/Documents/rocket-chip/emulator/verilator/install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -O1 -std=c++11 -I/home/tassneemhelal/Documents/rocket-chip//include -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/verilator.h -include /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.plusArgs -I/home/tassneemhelal/Documents/rocket-chip/emulator/generated-src -include /home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig/VTestHarness.h   -c -o SimDTM.o /home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimDTM.cc
/home/tassneemhelal/Documents/rocket-chip/src/main/resources/csrc/SimDTM.cc:3:23: fatal error: fesvr/dtm.h: No such file or directory
compilation terminated.
VTestHarness.mk:71: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/tassneemhelal/Documents/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig'
Makefrag-verilator:71: recipe for target 'emulator-freechips.rocketchip.system-DefaultConfig' failed
make: *** [emulator-freechips.rocketchip.system-DefaultConfig] Error 2

mayoru, 2018-05-24:
You don't have RISCV environment variable defined. Have you run riscv-tools/build.sh?  Find dtm.h in your $RISCV/include/fesvr directory.

tassneemhelal, 2018-05-24:
Yeah, I've done all of that and actually, the command was running successfully yesterday and now it gives me that error and I don't know why. @mayoru

mayoru, 2018-05-24:
You don't have RISCV environment variable defined:
echo $RISCV