chipsalliance/rocket-chip #1449: How do I attach different Rocc to different tile

aicr, 2018-05-24:
I have an implementation with 4 tiles and some Roccs.
My question is how do I attach a specific RoCC to a specific tile. I have already used all the 4 custom instructions and need to add custom functionality that needed in one tile but not in others and vice versa.

hcook, 2018-05-30:
Doing this will require a little fancy footwork with the `config` library, but is certainly possible. The power of the `config.Parameters` class is that the value of a field can depend on the value of another field. This recursion is accomplished using the companion functions `site`, `here` or `up`. In particular, `site` allows you to look up another field's value from the perspective of the place where you started the initial lookup (the "site of the lookup"). In your case, you can use this capability to specialize the `BuildRoCC` field's value based on what tile you are in.
```
... extends Config((site, here, up) => {
  case BuildRoCC => site(TileKey).hartId match {
    case 0: List((p: Parameters) => { <some rocc> }, ...)
    case 1: List((p: Parameters) => { <some other rocc> }, ...)
    ...
})
```
The value returned by ` site(TileKey).hartId` will determine which tile you were in, based on where you called `p(BuildRoCC)` (which each tile does). Matching on that value allows for specialization, and you could match on odd vs even ids or something as well.

aicr, 2018-06-13:
Cool , thanks I'll try that

aicr, 2018-07-05:
Thanks, that worked , only a syntax correction it shold be:
`case 0 => List((p: Parameters) => { <some rocc> }, ...)`
This issue can be closed.

closed 2018-07-05 by aicr