chipsalliance/rocket-chip #1454: Error during elaborate phase

heshamelmatary, 2018-05-28:
Hi,

I'm trying to build a custom project around the rocket chip. As I'm not experienced with Chisel/Scala, I get an error that I'm not sure how to debug/understand. I'd appreciate any hints on how to detect the what am I doing wrong and the file/code that's causing this error:

> [info] Running custom.system.Generator ...
> [info] [0.003] Elaborating design...
> [error] (run-main-0) java.lang.reflect.InvocationTargetException
> [error] java.lang.reflect.InvocationTargetException
> [error]         at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
> .
> .
> .
> Caused by: java.lang.IllegalArgumentException: requirement failed: system_bus_xbar.node (a nexus) has 3 required outputs and no possible inputs
> [error]         at scala.Predef$.require(Predef.scala:224)
> [error]         at freechips.rocketchip.diplomacy.MixedNexusNode.resolveStar(Nodes.scala:474)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.liftedTree1$1(Nodes.scala:266)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.x$25$lzycompute(Nodes.scala:251)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.x$25(Nodes.scala:250)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.oPortMapping$lzycompute(Nodes.scala:250)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.oPortMapping(Nodes.scala:250)
> [error]         at freechips.rocketchip.diplomacy.MixedNode.oPortMapping(Nodes.scala:217)
> [error]         at freechips.rocketchip.diplomacy.MixedNode$$anonfun$iPorts$1.apply(Nodes.scala:290)
> [error]         at freechips.rocketchip.diplomacy.MixedNode$$anonfun$iPorts$1.apply(Nodes.scala:289)
> .
> .
> .
>

mayoru, 2018-05-29:
What's your configuration? Do you have any of these ports: MasterAXI4MemPort, MasterAXI4MMIOPort, SlaveAXI4Port? Do you have DCache in a scratchpad mode? Try changing these parameters.

hcook, 2018-05-30:
Sounds like you managed to generate a design in which the `system_bus_xbar`, i.e. the primary system interconnect, has 3 slaves (the sys port, memory bus, periphery bus) and no masters. Masters are usually tiles than contain caches. Did you override `RocketTilesKey` in your `Config` class?