chipsalliance/rocket-chip #1461: The boom does not compile with the last rocket because of the decode.scala corrections

pentin-as, 2018-05-31:
rocket-chip version: 4269a221777d12f949f5603f3a5ee19f7c84d015

Error text:
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:51)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:48)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:297)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:295)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:295)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:93)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:54)
[error] 	at boom.system.Generator$.elaborate(Generator.scala:13)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:91)
[error] 	at boom.system.Generator$.circuit$lzycompute(Generator.scala:13)
[error] 	at boom.system.Generator$.circuit(Generator.scala:13)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:97)
[error] 	at boom.system.Generator$.generateFirrtl(Generator.scala:13)
[error] 	at boom.system.Generator$.delayedEndpoint$boom$system$Generator$1(Generator.scala:95)
[error] 	at boom.system.Generator$delayedInit$body.apply(Generator.scala:13)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:381)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at boom.system.Generator$.main(Generator.scala:13)
[error] 	at boom.system.Generator.main(Generator.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.OutOfMemoryError: GC overhead limit exceeded
[error] 	at scala.collection.mutable.ListBuffer.$plus$eq(ListBuffer.scala:174)
[error] 	at scala.collection.mutable.ListBuffer.$plus$eq(ListBuffer.scala:45)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] 	at scala.collection.immutable.List.foreach(List.scala:381)
[error] 	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
[error] 	at scala.collection.immutable.List.map(List.scala:285)
[error] 	at freechips.rocketchip.rocket.Simplify$$anonfun$21$$anonfun$apply$33.apply(Decode.scala:131)
[error] 	at freechips.rocketchip.rocket.Simplify$$anonfun$21$$anonfun$apply$33.apply(Decode.scala:131)
[error] 	at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[error] 	at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
[error] 	at scala.collection.immutable.List.foreach(List.scala:381)
[error] 	at scala.collection.TraversableLike$class.flatMap(TraversableLike.scala:252)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:344)
[error] 	at freechips.rocketchip.rocket.Simplify$$anonfun$21.apply(Decode.scala:131)
[error] 	at freechips.rocketchip.rocket.Simplify$$anonfun$21.apply(Decode.scala:131)
[error] 	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:84)
[error] 	at freechips.rocketchip.rocket.Simplify$.getCover(Decode.scala:131)
[error] 	at freechips.rocketchip.rocket.Simplify$.apply(Decode.scala:142)
[error] 	at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$7.apply(Decode.scala:39)
[error] 	at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$7.apply(Decode.scala:29)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
[error] 	at scala.collection.immutable.Range.foreach(Range.scala:166)
[error] 	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
[error] 	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error] 	at freechips.rocketchip.rocket.DecodeLogic$.apply(Decode.scala:29)
[error] 	at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$13.apply(Decode.scala:50)
[error] 	at freechips.rocketchip.rocket.DecodeLogic$$anonfun$apply$13.apply(Decode.scala:49)
[error] 	at scala.collection.TraversableLike$WithFilter$$anonfun$map$2.apply(TraversableLike.scala:728)
[error] 	at scala.collection.immutable.List.foreach(List.scala:381)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 210 s, completed May 31, 2018 4:55:30 PM

All is compiled if I cancel the following corrections:
https://github.com/freechipsproject/rocket-chip/commit/492ac75ecf2454859dc97be6b6473111df1586ff#diff-61210e44ff43865a122adf453037b4c6R102
https://github.com/freechipsproject/rocket-chip/commit/492ac75ecf2454859dc97be6b6473111df1586ff#diff-61210e44ff43865a122adf453037b4c6R130

pentin-as, 2018-06-01:
Yes you are right. Excuse me.

closed 2018-06-01 by pentin-as

et-ccelio, 2018-06-01:
[error] Caused by: java.lang.OutOfMemoryError: GC overhead limit exceeded

Hmmmm.