chipsalliance/rocket-chip #1467: Doc about Debug Module Interface

IC-Dream, 2018-06-04:
Hi, 

I have generated the verilog of rocket with defaultconfig, and noticed that it contains some debug interface, like, *_dmi_req_valid/ready/bits_addr/bits_data/bits_op/bits_resp. etc. So how could i use these interface, since i want to intergrate rocket to some Soc, Is there any other doc to explain these interface, and it is not clear enough to me in riscv-debug-spec v0.13. like, it doesn't declare the *bits_resp,or the valid/ready's behavior keep same as AMBA?

Thanks for your help

mwachs5, 2018-06-04:
If you add a DTM (see
https://github.com/freechipsproject/rocket-chip/blob/master/README.md#debug
)

Then the interface is JTAG and aligns with what is documented in the RISC-V
debug spec.

The raw DMI interface isn’t really documented anywhere at the moment, but
it’s a DecoupledIO (Ready/valid) interface with opcode, address, and data.

IC-Dream, 2018-06-05:
@mwachs5 Thanks for your help, another question:
From the riscv-debug-spec, the debugger can write/read the field of op, In my opinion, when the debugger read the field of op, which includes 4 status, it is just like the *_bits_resp for this access, right?
[image]

Thanks in advance

closed 2018-10-22 by IC-Dream