chipsalliance/rocket-chip #1469: Request to separate rocket-chip and tool-chains.

sequencer, 2018-06-04:
These days I have been struggling the port rocket-chip by sifive/freedom.
And I have some advice to this project, hoping these will not offend.

1. Is it possible to use sbt with maven to include `firrtl`, `hardfloat`, `torture` as packages, which will lead version control into a stable way.
2. Using GitHub release page to publish new stable version of rocket-chip, making it possible to give a change log.
3. Remove the tool-chain which has been already packaged in many Linux distributions like Arch and Debian.
4. Add Comments to each class or Functions, and use these comments to generate a readable documentation.

chshibo, 2018-10-11:
Agree on the 4th point. And it's rather hard to understand the internal logic and the codes without comments.

cousteaulecommandant, 2019-12-08:
I think the 4th point might deserve an issue on its own.  I've lately been struggling trying to follow Rocket's Chisel source code, and the lack of in-source documentation makes the task almost impossible.
A mere description of what each class does or means would highly improve the readability of the source code, and greatly help in the development of derived projects.

sequencer, 2019-12-09:
Emmm, This seems to be my first issue. really old now.
1. most of packages has its own version now.
2. git tag is used for version.
3. tool-chain is option now.
4. I'm writing a diplomacy documentation for SiFive China internal documentation. PR is welcomed in [diplomacy](https://github.com/cnrv/diplomacy)

closed 2019-12-09 by sequencer