chipsalliance/rocket-chip #1470: Step by step installation request

pa1ly, 2018-06-04:
Dear all,
I am unable to build the the rocket chip tool.
Following the read me fail has left me lot of errors and unable to build it (./build.sh just gives errors).
I am using Ubuntu 18.04 64-bit in virtual box.
Please give me a step by step guide where I install all the required files/dependencies so that i can complete the build process successfully.
Thank you.

tassneemhelal, 2018-06-05:
These are the steps that you should follow:

$ git clone https://github.com/ucb-bar/rocket-chip.git
$ cd rocket-chip
$ git submodule update --init
$ export RISCV=_/path/to/riscv/toolchain/installation_ (Make sure to replace the italic part which is the path with the path of your rocket-chip folder)
$ cd rocket-chip/riscv-tools
$ git submodule update --init --recursive
$ export RISCV=_/path/to/install/riscv/toolchain_ Make sure to replace the italic part which is the path with the path of your riscv-tools folder)
$ export MAKEFLAGS="$MAKEFLAGS -j2"
BEFORE THE NEXT STEP MAKE SURE THAT THE PERMISSION OF THE ROCKET-CHIP FOLDER IS SET TO DELETE AND ACCESS FOR EVERYONE
$ ./build.sh
$ cd ..
$ cd emulator
$ make
$ make -j2 run-asm-tests
$ make -j2 run-bmark-tests
$ make debug
$ cd ..
$ cd vsim
$ make verilog

If you face any problem while facing them just let me know. :)

tampler, 2018-06-09:
@tassneemhelal 
I can't follow your installation guide on Ubuntu 18.04 as well. I pulled and built the latest RISCV toolchain and made all exports. 
My javac version is 1.8.0_171.
When I cd emulator and run make -j2 I get the following error: 
mkdir -p /home/bku/work/cores/rocket-chip/emulator/generated-src/
cd /home/bku/work/cores/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/bku/work/cores/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/bku/work/cores/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/bku/work/cores/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/bku/work/cores/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running freechips.rocketchip.system.Generator /home/bku/work/cores/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig
[info] [0.005] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:52)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:49)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:297)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:295)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:295)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:93)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$class.elaborate(GeneratorUtils.scala:57)
[error] 	at freechips.rocketchip.system.Generator$.elaborate(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:94)
[error] 	at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:100)
[error] 	at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:88)
[error] 	at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at freechips.rocketchip.system.Generator$.main(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator.main(Generator.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.AbstractMethodError: freechips.rocketchip.system.ExampleRocketSystem.freechips$rocketchip$subsystem$HasTiles$_setter_$freechips$rocketchip$subsystem$HasTiles$$lookupByHartId_$eq(Lfreechips/rocketchip/tile/LookupByHartIdImpl;)V
[error] 	at freechips.rocketchip.subsystem.HasTiles$class.$init$(HasTiles.scala:29)
[error] 	at freechips.rocketchip.subsystem.RocketSubsystem.<init>(RocketSubsystem.scala:63)
[error] 	at freechips.rocketchip.system.ExampleRocketSystem.<init>(ExampleRocketSystem.scala:12)
[... cut ...]
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 5 s, completed Jun 9, 2018 8:01:09 PM
Makefrag-verilator:12: recipe for target '/home/bku/work/cores/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.fir' failed
make: *** [/home/bku/work/cores/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.fir] Error 1

Please help me to resolve this and run Verilator simulation. Thanks.
PS: I get similar error for BOOM default built. Am I missing something with your Chisel flow ?

LGKev, 2018-08-06:
@tampler I noticed you're using Ubuntu 18.04. I had similar issues with the sbt and java. My solution came after randomly trying ubuntu 16.04. My java version is 1.80_171, which should be the default for 16.04 ubuntu. At one point trying to get rocket chip to correctly build, I ended up uninstalling everything related to Java and installed the specific 1.8 version. I can't say for sure if this will help but I know that 18.04 definitively did not work for me. I am also running Virtual Box 5.2.12

RubelAhmed57, 2018-10-31:
@tassneemhelal Thank you very much. 
I followed your instruction: from 
$ git clone https://github.com/ucb-bar/rocket-chip.git
$ cd rocket-chip
 .
 .
 .  
$ cd vsim
$ make verilog

and found some folders like output, generated src etc. Now I want to know:

1. How to build a simulation model for this SoC? (or it is already generated in my steps?)
2. How to simulate the model?(out are my expected output/how can I be assured my installation is fine?)

angi10, 2019-10-23:
> These are the steps that you should follow:
>
> $ git clone https://github.com/ucb-bar/rocket-chip.git
> $ cd rocket-chip
> $ git submodule update --init
> $ export [...]

When i arrive at `cd emulator
make`

I have a problem: there is a first warning that the system doesn't find the submodule api-config-sifive, and i think this results in an error at the end of the process

closed 2020-02-09 by mwachs5