chipsalliance/rocket-chip #1479: Error while running make -j6 rum-asm-tests

pa1ly, 2018-06-08:
:: USE VERBOSE OR DEBUG MESSAGE LEVEL FOR MORE DETAILS
unresolved dependency: org.scala-sbt#sbt;1.1.1: not found
Error during sbt execution: Error retrieving required libraries
  (see /home/pavan/.sbt/boot/update.log for complete log)
Error: Could not retrieve sbt 1.1.1

**The update.log file had the following error:**

ERROR: Server access Error: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty url=https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/1.1.1/sbt-1.1.1.pom
        sbt-maven-releases: resource not reachable for org/scala-sbt#sbt;1.1.1: res=https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/1.1.1/sbt-1.1.1.pom
         trying https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/1.1.1/sbt-1.1.1.jar
                tried https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/1.1.1/sbt-1.1.1.jar

edcote, 2018-06-11:
Allow me to introduce you to StackOverflow for infra questions unrelated to the repo.

https://stackoverflow.com/questions/32181019/server-access-error-while-running-sbt-assembly

pa1ly, 2018-06-14:
I tried. It does not help.

pa1ly, 2018-06-14:
i tried to download sbt 1.1.1 and copy files from local-preloaded in the downloaded version to my the ~/.ivy2/local . This results in a series of errors and does not help me.

pa1ly, 2018-06-16:
got it, changed OS to fedora 28

closed 2020-02-09 by mwachs5