chipsalliance/rocket-chip #1486: Problem about dumping VCD waveforms

IC-Dream, 2018-06-12:
Hi,
when i make -j4 run-bmark-tests-debug to dump VCD waveforms, it got stucked for dozen hours as follow:

make[1]: Leaving directory `/cm/freedom/rocket-chip/emulator/generated-src-debug/freechips.rocketchip.system.DefaultConfig'
rm -rf output/mm.riscv.vpd.vcd && mkfifo output/mm.riscv.vpd.vcd
rm -rf output/spmv.riscv.vpd.vcd && mkfifo output/spmv.riscv.vpd.vcd
rm -rf output/mt-vvadd.riscv.vpd.vcd && mkfifo output/mt-vvadd.riscv.vpd.vcd
rm -rf output/median.riscv.vpd.vcd && mkfifo output/median.riscv.vpd.vcd
vcd2vpd output/spmv.riscv.vpd.vcd output/spmv.riscv.vpd > /dev/null &
vcd2vpd output/mm.riscv.vpd.vcd output/mm.riscv.vpd > /dev/null &
vcd2vpd output/mt-vvadd.riscv.vpd.vcd output/mt-vvadd.riscv.vpd > /dev/null &
/bin/bash: vcd2vpd: command not found
vcd2vpd output/median.riscv.vpd.vcd output/median.riscv.vpd > /dev/null &
/bin/bash: vcd2vpd: command not found
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/spmv.riscv.vpd.vcd output/spmv.riscv 3>&1 1>&2 2>&3 | /path/to/install/riscv/toolchain/bin/spike-dasm  > output/spmv.riscv.out && [ $PIPESTATUS -eq 0 ]
/bin/bash: vcd2vpd: command not found
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/mm.riscv.vpd.vcd output/mm.riscv 3>&1 1>&2 2>&3 | /path/to/install/riscv/toolchain/bin/spike-dasm  > output/mm.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/mt-vvadd.riscv.vpd.vcd output/mt-vvadd.riscv 3>&1 1>&2 2>&3 | /path/to/install/riscv/toolchain/bin/spike-dasm  > output/mt-vvadd.riscv.out && [ $PIPESTATUS -eq 0 ]
/bin/bash: vcd2vpd: command not found
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/median.riscv.vpd.vcd output/median.riscv 3>&1 1>&2 2>&3 | /path/to/install/riscv/toolchain/bin/spike-dasm  > output/median.riscv.out && [ $PIPESTATUS -eq 0 ]

@seldridge, I noticed that you did some comment about the following issue, like the below:

https://github.com/freechipsproject/rocket-chip/issues/1285

"FWIW, @clx7913: you don't need to run these tests. If you want to generate a waveform, you can use the +verbose option when running one of these programs."

And i tried some modification in ~/emulator/Makefile based on my understanding about your comment, like that:  add "-v$@.vcd" in the below command

$(output_dir)/%.out: $(output_dir)/% $(emu)
	./$(emu) +max-cycles=$(timeout_cycles) +verbose -v$@.vcd $< $(disasm) $@ && [ $$PIPESTATUS -eq 0 ]

But it didn't work. it seems that i didn't figure out what you mean ? or there are some other issues?

Thanks in advance

seldridge, 2018-06-12:
There's a mistake in that comment. `+verbose` turns on Chisel printfs and `-v<FILE>` will dump to a VCD file. 

However, there's a fundamental question (in the other issue) of why you want to run all these tests and to dump complete waveforms while running them. This will be _a lot_ of very, very large VCD files. Usually, only a build server will ever have some reason to run these.

If you do intend to do this, then you will need to modify (ideally) [`RocketTestSuite.scala`](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/RocketTestSuite.scala)  so that it emits a suitable Makefile fragment (`generated-src/freechips.rocketchip.system.DefaultConfig.d`) that is not trying to generate `*.vpd` file and only generates `*.vcd` files.

Do note that you do not need to run these tests _and dump full waveforms_ to validate that your design works.

IC-Dream, 2018-06-13:
@seldridge Yes, I only want to run dhrystone benchmark and dump VCD waveform, so what modification I  need to do ?

Thanks so much

seldridge, 2018-06-13:
When you build either the C simulator or VCS simulator, you then have a binary that will allow you to run programs directly on a simulated Rocket Chip.

E.g., the following will build a C simulator capable of dumping a waveform:
```
cd emulator
make debug
```

That produces a binary, `emulator-freechips.rocketchip.system-DefaultConfig`. This has ample help text that should help clarify what you can pass to this. However, given a bare metal binary, `myBinary`, you can run this and dump a waveform with:

```
./emulator-freechips.rocketchip.system-DefaultConfig -v myWaveform.vcd myBinary
```

When you run `make -j4 run-bmark-tests-debug`, this is running that binary multiple times with different options and target binaries.

IC-Dream, 2018-06-14:
@seldridge I substituted the “”vpd“” with “”vcd“” and modified the val benchmarks and rv32udBenchmarks in the RocketTestSuite.scala, it also works.  

Thank you all the same

closed 2018-06-14 by IC-Dream

HaFred, 2019-11-27:
> rv32udBenchmarks

Hi, may I ask you how to modify val benchmarks specifically?

HaFred, 2019-11-27:
> When you build either the C simulator or VCS simulator, you then have a binary that will allow you to run programs directly on a simulated Rocket Chip.
>
> E.g., the following will build a C simulator [...]

I got invalid option '--v' for the default config running