chipsalliance/rocket-chip #1487: how to test rocket-chip

sharinoz, 2018-06-12:
hi
I am not quiet sure that I managed to install the rocket chip successfully
how can I run a simple test, something such as "hello world!" that once its working I can be sure that everything is installed and in place?

thanks

CLappin, 2018-06-12:
Hi
Have you rub the example c code from the readme?
Thanks,
Ciaran

CLappin, 2018-06-12:
sure, have a look at this section "How should I use the Rocket chip generator?"  in the readme. it tells you about asm file that can be run using the cycle accurate simulator as well as other test methods.

sharinoz, 2018-06-13:
hey, I just did it
now I have an output directory at rocket-chip/emulator with a lot of files
is this what suppose to happen? down it mean that the install has done successfully?

CLappin, 2018-06-15:
Hi  @sharinoz ,

yes, if you followed the first step you should have a lot of files in the emulator directory. 
From the README.
The output of the executed assembly tests and benchmarks can be found at emulator/output/*.out. Each file has a cycle-by-cycle dump of write-back stage of the pipeline. Here's an excerpt of emulator/output/rv64ui-p-add.out:

C0: 483 [1] pc=[00000002138] W[r 3=000000007fff7fff][1] R[r 1=000000007fffffff] R[r 2=ffffffffffff8000] inst=[002081b3] add s1, ra, s0
C0: 484 [1] pc=[0000000213c] W[r29=000000007fff8000][1] R[r31=ffffffff80007ffe] R[r31=0000000000000005] inst=[7fff8eb7] lui t3, 0x7fff8
C0: 485 [0] pc=[00000002140] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000000] unknown
The first [1] at cycle 483, core 0, shows that there's a valid instruction at PC 0x2138 in the writeback stage, which is 0x002081b3 (add s1, ra, s0). The second [1] tells us that the register file is writing r3 with the corresponding value 0x7fff7fff. When the add instruction was in the decode stage, the pipeline had read r1 and r2 with the corresponding values next to it. Similarly at cycle 484, there's a valid instruction (lui instruction) at PC 0x213c in the writeback stage. At cycle 485, there isn't a valid instruction in the writeback stage, perhaps, because of a instruction cache miss at PC 0x2140.

Thanks 
Ciaran

sharinoz, 2018-06-19:
hi @CLappin 
thanks for the detailed answer

closed 2020-02-09 by mwachs5