chipsalliance/rocket-chip #1492: Issue about issues

edcote, 2018-06-13:
FWIW. This forum is full of what I judge to be non-issues.  A quick count of outstanding issues shows that the majority of them are in fact requests for help from new users.

Other than inbox spam, I am not personally affected.  My opinion is that users will be better served by asking these questions elsewhere.

My request is to adopt the strategy used the chisel3 repo (i.e. questions need to be asked before submitting issue)

https://github.com/freechipsproject/chisel3/issues/new

I can offer to make the change myself but need admin rights.  With admin blessing, I can also force close these non-issue issues, asking the user to move the question to StackOverflow.

seldridge, 2018-06-13:
Interestingly, I started to write up an RFC related to this. However, I would urge that we need to keep this civil and friendly. One current problem is that there exist known forums/groups for Chisel/FIRRTL, but not for Rocket Chip. E.g., it's unclear (to me, at least) where to redirect people for technical support.

Anyway, thanks for bringing this up @edcote. This is an issue, particularly given the size of the project. It likely makes sense to treat this as an RFC going forward.

FWIW, here's what I had started to write up:

> While this is an incredibly good thing and speaks to the reach of Rocket Chip... there are a plethora of issues filed on Rocket Chip and related repositories (excepting Chisel and FIRRTL) that blur the line between actual issues (e.g, the README has incorrect instructions), technical support (e.g., "How do I do `$MyResearch`?"), and fundamental questions related to Scala, sbt, or Chisel that may be better directed elsewhere?

> #### Questions:
>  - Is the current state of the world problematic and something that should be changed?
>  - What is in scope for issues filed on Rocket Chip?
>  - Would an issue template help clarify what is in scope and what should this look like?
>  - If the scope of reasonable issues limits a large fraction of classes of questions, what would be a good forum for these non-issue questions? Possibilities include: SiFive forums, Stack Overflow, a "Rocket Users" Google Group / mailing list, IRC, etc.
>  - How can this be done in an inclusive/friendly way that makes Rocket Chip accessible to new users, e.g., _not_ "RTFC/GTFO SO" -> `close issue`?

edcote, 2018-06-17:
I am going to close this "issue" and contact you @seldridge offline.  Kudos on your political correctness, btw ;)

closed 2018-06-17 by edcote

colinschmidt, 2018-07-18:
FWIW there is now a stack overfow tag https://stackoverflow.com/questions/tagged/rocket-chip