chipsalliance/rocket-chip #1495: Rocket chip simulation shows unexpected instruction count

radiosonde, 2018-06-14:
This is a replica of my Stackoverflow post (https://stackoverflow.com/questions/50847440/rocket-chip-simulation-shows-unexpected-instruction-count).

The following two code snippets differ only the value loaded into the x23 register, but the instruction counts shown by `minstret` (reported by a Verilator simulation of the Rocket chip) differ substantially.  Is this a bug, or am I doing something wrong?

The `read_csr()` function is from the RISC-V Frontend Server Library (https://github.com/riscv/riscv-fesvr/blob/master/fesvr/encoding.h), and the rest of the code [syscalls.c, crt.S, test.ld] is similar to the RISC-V benchmarks (https://github.com/riscv/riscv-tests/tree/master/benchmarks/common).

I have checked that the relevant portions of the two binaries have the same instructions in both binaries, except for the difference in initialization of the x23 register.

Dividing 0x0fffffff by 0xff, repeating 1024 times: **3260 instructions**.

	size_t instrs = 0 - read_csr(minstret);

	asm volatile (
			"mv             x20,    zero;"
			"li             x21,    1024;"
			"li             x22, 	0xfffffff;"
			"li             x23, 	0xff;"

		"loop:"
			"div 		    x24,  x22,  x23;"
			"addi           x20,  x20,  1;"
			"bleu           x20,  x21,  loop;"

		::: "x20", "x21", "x22", "x23", "x24", "cc"
	);

	instrs += read_csr(minstret);

Dividing 0x0fffffff by 0xffff, repeating 1024 times: **3083 instructions**.

	size_t instrs = 0 - read_csr(minstret);

	asm volatile (
			"mv             x20,    zero;"
			"li             x21,    1024;"
			"li             x22, 	0xfffffff;"
			"li             x23, 	0xffff;"

		"loop:"
			"div 		    x24,  x22,  x23;"
			"addi           x20,  x20,  1;"
			"bleu           x20,  x21,  loop;"

		::: "x20", "x21", "x22", "x23", "x24", "cc"
	);

	instrs += read_csr(minstret);

Here, 3083 instructions seems correct (1024 * 3 = 3072).  Since `minstret` counts retired instructions, it seems strange that first example executed ~200 more instructions.  These results are always the same no matter how many times I run these two programs.

aswaterman, 2018-06-14:
It's because a debug interrupt occurred during one sequence and not the other.  The debug unit caused some more instructions to execute on the processor.

I realize that's not ideal for benchmarking.

radiosonde, 2018-06-14:
Thanks! How can I get to know when the debug interrupt did (or did not) fire, and what do you think caused the debug interrupt?

aswaterman, 2018-06-14:
If you look at the execution log, you'll see the PC change to address 0x800 - that's the debug address range.

The debug interrupt happened because the debug port is being used to poll whether the benchmark has finished running.

radiosonde, 2018-06-14:
Thanks much! The debug address range showed up at different times during the execution of the two binaries. I reduced the loop count, and I now see roughly consistent instruction counts across the executions. Closing this issue.

closed 2018-06-14 by radiosonde

sammy17, 2021-09-19:
I am also seeing many debug instructions when I try to emulate even a simple program (riscv-tests rv64ua-v-amoxor_d). Many debug instructions were executed before and in between the program instructions. 
Is there a way to disable these debug interrupts in the emulator?

I am trying to run a longer program as well which never seems to end due to many debug routines that seem to get executed in between the program instructions. 

Please let me know if there is any way to disable these interrupts. 

I am using latest rocket-chip as of 2021 Sept 18 with default configuration.