chipsalliance/rocket-chip #1506: errors while run CONFIG=DefaultConfigRBB make debug in ../emulator

sharinoz, 2018-06-19:
Hi,
I am trying to debug the rocket core, but in the very beginning I am getting a lot of errors
 

rotemshahar@asic2-serv3:~/rocket-chip/riscv-tools$ cd ../emulator/
rotemshahar@asic2-serv3:~/rocket-chip/emulator$ CONFIG=DefaultConfigRBB make debug
mkdir -p /home/rotemshahar/rocket-chip/emulator/generated-src/
cd /home/rotemshahar/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/rotemshahar/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/rotemshahar/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfigRBB"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/rotemshahar/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons:
[info] Set current project to rocketchip (in build file:/home/rotemshahar/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running freechips.rocketchip.system.Generator /home/rotemshahar/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfigRBB
[error] (run-main-0) java.lang.Exception: Unable to find part "freechips.rocketchip.system.DefaultConfigRBB" from "ArrayBuffer(freechips.rocketchip.system.DefaultConfigRBB)", did you misspell it?
[error] java.lang.Exception: Unable to find part "freechips.rocketchip.system.DefaultConfigRBB" from "ArrayBuffer(freechips.rocketchip.system.DefaultConfigRBB)", did you misspell it?
[error]         at Chisel.package$throwException$.apply(compatibility.scala:387)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:38)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:33)
[error]         at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
[error]         at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
[error]         at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:33)
[error]         at freechips.rocketchip.system.Generator$.getConfig(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp$class.config(GeneratorUtils.scala:92)
[error]         at freechips.rocketchip.system.Generator$.config$lzycompute(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.config(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp$class.params(GeneratorUtils.scala:93)
[error]         at freechips.rocketchip.system.Generator$.params$lzycompute(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.params(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:94)
[error]         at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:100)
[error]         at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:88)
[error]         at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
[error]         at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error]         at scala.App$class.main(App.scala:76)
[error]         at freechips.rocketchip.system.Generator$.main(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator.main(Generator.scala)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.ClassNotFoundException: freechips.rocketchip.system.DefaultConfigRBB
[error]         at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
[error]         at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
[error]         at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
[error]         at java.lang.Class.forName0(Native Method)
[error]         at java.lang.Class.forName(Class.java:264)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:35)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:33)
[... 35 lines cut ...]
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 4 s, completed Jun 19, 2018 12:01:39 PM
Makefrag-verilator:12: recipe for target '/home/rotemshahar/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfigRBB.fir' failed
make: *** [/home/rotemshahar/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfigRBB.fir] Error 1

I did again all the steps to keep everything up-to-date and also did again /riscv-tools/build.sh but it did not helped
I am not sure how to fix this problem
thanks
Shahar

mwachs5, 2018-06-19:
There is no such config DefaultConfigRBB. Did you create one?

sharinoz, 2018-06-24:
@mwachs5 no I didn't create one, ho do I do that?

MeibaoXu, 2018-06-25:
Hi @sharinoz 

add 
`class DefaultConfigRBB extends Config(
new WithJtagDTMSystem ++ new WithNBigCores(1) ++ new BaseConfig)`

to the end of the [this chisel file](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/Configs.scala)

sharinoz, 2018-08-20:
hi @MeibaoXu 
maybe you can help me
i managed to solve this problem thanks to you, but i still can't debug properly
each time i start the GDB and try to load, it seems like it's try loading but then always fails:

rotemshahar@asic2-serv3:~$ riscv64-unknown-elf-gdb riscv-isa-sim//rot13-64
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from riscv-isa-sim//rot13-64...done.
(gdb) set remotetimeout 2000
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) load
Loading section .text, size 0x74 lma 0x10010000
Loading section .data, size 0x22 lma 0x10010078
Load failed
(gdb) print wait
Cannot access memory at address 0x1001009c
(gdb) q
A debugging session is active.

        Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/rotemshahar/riscv-isa-sim/rot13-64, Remote target
Ending remote debugging.

can you help me undestand why is this happening?
thanks
Shahar