chipsalliance/rocket-chip #1515: RoccExampleConfig throwing java.lang.IllegalAccessError

seldridge, 2018-06-22:
On master with `RoccExampleConfig` I'm getting a `java.lang.IllegalAccessError` tied to [line 91 of LazyRoCC.scala](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/tile/LazyRoCC.scala#L91) which doesn't seem to make sense. This seems to be trying to access some reflection-related member in `RocketTileModuleImp` from `LazyRoCC` and failing, but I'm having some trouble understanding exactly what is going on here.

```
[error] Caused by: java.lang.IllegalAccessError: tried to access field freechips.rocketchip.tile.RocketTileModuleImp.reflPoly$Cache4 from class freechips.rocketchip.tile.HasLazyRoCCModule$class
[error] 	at freechips.rocketchip.tile.HasLazyRoCCModule$class.reflMethod$Method4(LazyRoCC.scala:91)
```

To reproduce this:

```bash
make CONFIG=RoccExampleConfig
```

Somewhat strangely, initial triage of this, like commenting out line 91 and then uncommenting it result in this switching to a `NullPointerException` tied to [line 105](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/tile/LazyRoCC.scala#L105). (A stateful error? :fearful:)

h/t @aroelke

colinschmidt, 2018-06-23:
Does this present on a clean repo? Sometimes strange stateful errors crop up with bad sbt caching (i.e. try deleting target et al)

seldridge, 2018-06-23:
Good question... I tested this on a clean clone and got this behavior. For some added sanity on the environment, I blew away my local `~/.ivy2` just in case that was contributing.

A full stack trace is available here: https://gist.github.com/seldridge/b5ebcd2e3a0357a93ccd83a7df3f4289

seldridge, 2018-06-23:
After bisecting to check if Rocket gets past elaboration, c563c9ce83af70dfbd019aa1701d71b4ac60e700 appears to be the breaking commit.

aswaterman, 2018-06-23:
It's not obvious to me why that broke it.  LMK if you figure it out...

seldridge, 2018-06-23:
@aswaterman: I'm glad I'm not the only one!

I'm still lost on the specific reflection error. Sometimes this chooses to manifest as a complaint about `fpuOpt` being `null` (which it is when the `HasLazyRoCCModule` constructor is running). I can get around this with a `roccInit` method or something, but that leads back to the reflection error.

I'll pull on it and see where I get.

I will also (finally) push a PR to just build `RoccExampleConfig` in Travis once this error is corrected.

PR #1516 (2018-06-27): RoCC Build Fixes
This fixes a problem where RoCC-enabled tiles would fail to build. This was due to bad inheritance where the inheriting class defined the FPU before it was used. The connections are now made more explicit by moving some code from the RoCC into the Tile.

- Creates a new `HasFpuOpt` trait that instantiates the FPU
- Mixes in the `HasFpuOpt` trait _before_ the `HasLazyRoCCModule` trait to make the `fpuOpt` member available for use inside the RoCC implementation
- Removes a kludge where the RoCC would first connect to a wire and then the core would connect to the same wire (assum [... cut]
closed 2018-06-27 by hcook