chipsalliance/rocket-chip #1523: Timing get worse ???

IC-Dream, 2018-06-28:
Hi,

I synthesize the rocket core based on 28nm processing and also substitute the behavior SRAMs with the vendor-specific SRAMs, got the following timing report:
[image]

that's is to say, the rocket core could run at most 500Mhz. Compared to the official data(page 15), there is a big gap.  According to the official data, Based on 40nm processing, the rocket core could run over 1Ghz. 

https://www.hotchips.org/wp-content/uploads/hc_archives/hc27/HC27.25-Tuesday-Epub/HC27.25.70-Processors-Epub/HC27.25.731-RISC-V-Lee-UCB-V4.with-backup.pdf

Why does the timing get worse ? or Have you did some optimization, but didn't release on github?

Any help will be appreciated.

edcote, 2018-06-28:
Did you apply retiming to the FPU?

IC-Dream, 2018-08-16:
I checked the generated verilog, for FMAC, it directly use the "*" operator, Doesn't the rocket generator implement the FMAC?

[image]

Thanks
Dream