chipsalliance/rocket-chip #1528: Emulating BlackBox doew not work

Ssavas, 2018-07-05:
Hi,

I am using BlackBox with HasBlackBoxInline to implement a module with a few inputs and outputs. I inline the verilog code with setInline(). However, while generating the emulator (in emulator directory) I get a compilation error which can be seen below. I thought emulator supports black boxes, am I wrong? (I am using the rocket-chip within fpga-zynq repo) (https://github.com/ucb-bar/fpga-zynq)

> make[1]: *** No rule to make target `NoCRouter.v', needed by `/home/suleyman/Desktop/Suleyman/fpga-zynq/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DualCoreSmallConfig/VTestHarness.h'.  Stop.

Here is the code I use:

`class NoCRouter extends BlackBox with HasBlackBoxInline{
	val io = IO(new Bundle {
		// some io pins 		
	}
)

//	setResource("../testchipip/vsrc/NoCRouter.v")
//	setResource("/NoCRouter.v")
setInline("NoCRouter.v", 
s"""
	|module NoCRouter(
	|	-//some io pins
	|);
	|endmodule
""".stripMargin)
}`

Ssavas, 2018-07-09:
I had a source file called NoCRouter.v in the wrong directory. Removing it fixed the problem.

Ssavas, 2018-07-09:
It seems like I was wrong. The error is still there.

I tried using HasBlackBoxResource and provide the source verilog file. I couldn't figure out where to put the verilog source file (in the file system). Then I removed `BlackBoxSourceHelper.copyResourceToFile(resourceId,outFile)`
from the BlackBoxSourceHelper.scala and copied the file myself into the emulator folder. It seemed to compile the emulator just fine but I didn't test it. 

Now the question is, where should I place the verilog source? I tried rocket-chip/src/main/resources but it didn't work.

hcook, 2018-07-10:
`rocket-chip/src/main/resources/vsrc`?

closed 2018-07-31 by hcook