chipsalliance/rocket-chip #1534: How do I generate just a part of ROM or Something?

TonyFwany, 2018-07-10:
Hi. I'm tony.
I try to generate ROM Blackbox for changing this part what I want. 
I 'm almost done about controller design and codes.

So I think I need to do verification my design part. 
But it's not easy. I know to do verification scope all the part about Generated Rocket Chip. 
But I want to do just ROM.

So Could I ask something about generation just a part like rom, tilelink, and Cores?

I try to do blocking on Top file about "with SMT". 
But I touch DontTouch and HasRTCModuleImp, it gave me error.

TonyFwany, 2018-07-10:
I don't want to get it out on generated  verilog code.

and If I'm wrong about sentences, I'm sorry and I'm not a good in English.

michael-etzkorn, 2022-10-15:
I'll close this as a repeat of #3150

closed 2022-10-15 by michael-etzkorn