chipsalliance/rocket-chip #1543: bad logic synth results

korelin-dima, 2018-07-18:
Hello World!
A few days ago i have tried to synthesize Rocket with DefaultConfig(used commit 5d7a0e7c4227) using 90nm library. Cache was generated as  registra block from data_arrays_0_0_ext.
I've aimed to 200 MHz (C2C - 5ns) and got critical path that i can't solve.

         Path
        
        849: VIOLATED (-5785 ps) Setup Check with Pin ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/ram_reg_187_92/ck->d
                  Group: C2C
             Startpoint: (R) ExampleRocketSystem_inst/tile/frontend/icache/s2_tag_hit_0_reg/ck
                  Clock: (R) clock
               Endpoint: (R) ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/ram_reg_187_92/d
                  Clock: (R) clock

                     Capture       Launch     
        Clock Edge:+    5000            0     
       Src Latency:+       0            0     
       Net Latency:+       0 (I)        0 (I) 
           Arrival:=    5000            0     
                                              
             Setup:-     162                  
     Required Time:=    4838                  
      Launch Clock:-       0                  
         Data Path:-   10623                  
             Slack:=   -5785                  

                                         Timing Point                                            Flags   Arc   Edge   Cell      Fanout  Load Trans Delay Arrival 
                                                                                                                                        (fF)  (ps)  (ps)   (ps)  

          ExampleRocketSystem_inst/tile/frontend/icache/s2_tag_hit_0_reg/ck                                -       -     R     (arrival)  317686     -     0     -       0 
          ExampleRocketSystem_inst/tile/frontend/icache/s2_tag_hit_0_reg/q                                 -       ck->q R     dffpq_4        12 301.2   186   642     642 
          ExampleRocketSystem_inst/tile/frontend/icache/p0030A/x                                           -       a->x  F     inv_4           5 249.5   159   176     817 
          ExampleRocketSystem_inst/tile/frontend/icache/p0044A/x                                           -       a->x  R     inv_8          18 341.6    82    87     904 
          ExampleRocketSystem_inst/tile/frontend/icache/p0023A2457/x                                       -       a->x  F     aoi22_1         1  38.7   236   240    1144 
          ExampleRocketSystem_inst/tile/frontend/icache/p0024A/x                                           -       a->x  R     nand2_2        11 224.3   181   195    1339 
          ExampleRocketSystem_inst/tile/frontend/p0003A29/x                                                -       a->x  F     inv_2           4 134.1   168   186    1525 
          ExampleRocketSystem_inst/tile/frontend/p0004A/x                                                  -       c->x  R     nor3_4         12 345.7   310   284    1808 
          ExampleRocketSystem_inst/tile/frontend/p0017A10205/x                                             -       a->x  F     inv_4          16 337.9   211   244    2052 
          ExampleRocketSystem_inst/tile/frontend/n0144D/x                                                  -       a->x  R     nand2_1         1  18.9    71    92    2145 
          ExampleRocketSystem_inst/tile/frontend/inst_10263/x                                              -       b->x  F     nor4_1          3  57.2   135   265    2410 
          ExampleRocketSystem_inst/tile/frontend/inst_2/x                                                  -       a->x  R     nor4_2          2  38.1    69   330    2740 
          ExampleRocketSystem_inst/tile/frontend/inst_10264/x                                              -       b->x  F     oai222_1        1  18.8   301   258    2998 
          ExampleRocketSystem_inst/tile/frontend/fq/fopt18394/x                                            -       a->x  R     inv_1           2  38.1   103   119    3117 
          ExampleRocketSystem_inst/tile/frontend/fq/n0300D20123/x                                          -       b->x  F     oai21_1         2  37.9   207   188    3304 
          ExampleRocketSystem_inst/tile/core/ibuf/inst_6103/x                                              -       a->x  R     aoi21_1         1  19.3    85   129    3433 
          ExampleRocketSystem_inst/tile/core/ibuf/inst_6102/x                                              -       c->x  F     oai21_1         3 116.4   488   439    3873 
          ExampleRocketSystem_inst/tile/core/inst_107253/x                                                 -       b->x  R     oai21_4         1  37.8   113   122    3994 
          ExampleRocketSystem_inst/tile/core/inst_107252/x                                                 -       a->x  F     inv_2           1  19.0    44    74    4068 
          ExampleRocketSystem_inst/tile/core/p0029A202/x                                                   -       a->x  R     nand2_1         1  18.6    61    51    4120 
          ExampleRocketSystem_inst/tile/core/inst_107417/x                                                 -       b->x  F     nor3_1          1  19.0    93   100    4220 
          ExampleRocketSystem_inst/tile/core/inst_107416/x                                                 -       a->x  F     and2_1          1  18.8    56   132    4352 
          ExampleRocketSystem_inst/tile/core/n0342D/x                                                      -       c->x  R     oai21_1         1  18.5   106    67    4419 
          ExampleRocketSystem_inst/tile/core/p0000A44/x                                                    -       c->x  R     or4_2           2  55.9    75   210    4629 
          ExampleRocketSystem_inst/tile/core/p0001A106609/x                                                -       a->x  F     nor2_2          1  75.8   112   129    4758 
          ExampleRocketSystem_inst/tile/core/p0000A106568/x                                                -       c->x  R     oai22_4         1  77.6   146   120    4878 
          ExampleRocketSystem_inst/tile/core/p0000A107428/x                                                -       b->x  R     or2_8           1  18.8    32   162    5039 
          ExampleRocketSystem_inst/tile/core/drc_bufs2/x                                                   -       a->x  F     inv_1           2 115.0   250   214    5253 
          ExampleRocketSystem_inst/tile/core/drc_bufs107427/x                                              -       a->x  R     inv_4          16 355.8   144   160    5414 
          ExampleRocketSystem_inst/tile/frontend/Fn0407D11045/x                                            -       a->x  F     inv_2          15 361.1   388   358    5772 
          ExampleRocketSystem_inst/tile/frontend/inst_9807/x                                               -       a->x  R     nand2_1         1  19.0   100   115    5886 
          ExampleRocketSystem_inst/tile/frontend/p0383A/x                                                  -       b->x  F     nor2_1          1  19.0    79    93    5980 
          ExampleRocketSystem_inst/tile/frontend/p0384A10204/x                                             -       c->x  R     aoi21_1         1  19.3    89    92    6072 
          ExampleRocketSystem_inst/tile/frontend/p0026A10219/x                                             -       c->x  F     oai21_1         3  52.0   262   268    6340 
          ExampleRocketSystem_inst/tile/frontend/icache/inst_2339/x                                        -       d0->x F     mux2_2         12 340.7   378   531    6871 
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/drc_bufs540884/x -       a->x  R     inv_2           2 154.4   149   172    7043 
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/drc_bufs540889/x -       a->x  F     inv_4           6 304.4   183   189    7233 
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/Fn0290D540891/x  -       a->x  R     inv_4           6 168.7    85    93    7326 
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/n0289DT/x        -       b->x  F     nor3_2          2  96.2   151   155    7481 
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/Fn0277D207252/x  -       a->x  R     inv_4           6 191.2    88    95    7576 
[... cut ...]
          ExampleRocketSystem_inst/tile/frontend/icache/data_arrays_0/data_arrays_0_0_ext/ram_reg_187_92/d -       -     R     dffpq_2         1     -     -     0   10623 

Other paths you can find in file below

[rocket_critical_path.txt](https://github.com/freechipsproject/rocket-chip/files/2204494/rocket_critical_path.txt)

Timing that i've got is far below 40FO4 that i've aimed. Can you suggest some solutions? Maybe i haven't switched some configurations? 

P.S. previously i had problems with FPU, but coped with them by configuring delays.

jerryhethatday, 2020-12-31:
did you solve your problem?

csgxiong, 2023-06-19:
Any updates now ?

sequencer, 2023-06-19:
Closing stale issues. ext memory should be replaced by SRAM IP.
The rocket-chip project doesn't provide any support to specific ASIC flows for now, since they are always NDA part of this industry.

closed 2023-06-19 by sequencer