chipsalliance/rocket-chip #1554: How do I generate multicore configurations with scratchpad?

Ssavas, 2018-07-26:
I tried generating a multicore (dualcore) system with scratchpad with the following class 
```
class WithNSmallCoresScratchpad(n: Int) extends Config((site, here, up) => {
  case RocketTilesKey => {
    val small = RocketTileParams(
      core = RocketCoreParams(useVM = false, fpu = None),
      btb = None,
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 256,	// 16Kb scratchpad
        nWays = 1,
        nTLBEntries = 4,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes),
	scratch = Some(0x80000000L))),	// this doesn't work
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBEntries = 4,
        blockBytes = site(CacheBlockBytes))))
    List.tabulate(n)(i => small.copy(hartId = i))
  }
})
```

but I get an address overlapping error as follows:

> at chisel3.Driver$.elaborate(Driver.scala:93)
> Caused by: java.lang.IllegalArgumentException: requirement failed: Ports cannot overlap: AddressSet(0x80000000, 0x3fff) AddressSet(0x80000000, 0xfffffff)
> 	at scala.Predef$.require(Predef.scala:224)
> 	at freechips.rocketchip.diplomacy.AddressDecoder$$anonfun$apply$1$$anonfun$apply$2$$anonfun$apply$3.apply(AddressDecoder.scala:30)

The class I use is just a copy of WithNSmallCores class with 2 modifications (nSets = 256 and scratch = Some(0x80000000L))

Any ideas on how to have a multicore system with scratchpads instead of caches?

Thanks,

hcook, 2018-07-31:
You have to change the address range of each scratchpad so they don't conflict with each other. Otherwise tilelink doesn't know where to route messages to those addresses.

Ssavas, 2018-07-31:
Thanks for the answer. I have given different starting addresses to the scratchpads as seen below. However I still get the same error. It seems like one of the scratchpads takes the memory space from 0x80000000 to 0x90000000 and I don't know how to fix it. What else do I need to modify (and where would it be)?

 ```
 val small0 = RocketTileParams(
      hartId = 0,
      core = RocketCoreParams(useVM = false, fpu = None),
      btb = None,
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 256,	// 16Kb scratchpad
        nWays = 1,
        nTLBEntries = 4,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes),
		scratch = Some(0x80000000L))),
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBEntries = 4,
        blockBytes = site(CacheBlockBytes))))

    val small1 = RocketTileParams(
      hartId = 1,
      core = RocketCoreParams(useVM = false, fpu = None),
      btb = None,
      dcache = Some(DCacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 256,	// 16Kb scratchpad
        nWays = 1,
        nTLBEntries = 4,
        nMSHRs = 0,
        blockBytes = site(CacheBlockBytes),
		scratch = Some(0x80100000L))),
      icache = Some(ICacheParams(
        rowBits = site(SystemBusKey).beatBits,
        nSets = 64,
        nWays = 1,
        nTLBEntries = 4,
        blockBytes = site(CacheBlockBytes))))
```

Ssavas, 2018-08-03:
It seems that the overlap is between scratchpad address space and external memory address space in BaseConfig which can be seen below:

```
class BaseConfig extends Config(new BaseSubsystemConfig().alter((site,here,up) => {
  // DTS descriptive parameters
  case DTSModel => "freechips,rocketchip-unknown"
  case DTSCompat => Nil
  case DTSTimebase => BigInt(1000000) // 1 MHz
  // External port parameters
  case NExtTopInterrupts => 2
  case ExtMem => MasterPortParams(
                      base = x"8000_0000",
                      size = x"1000_0000", // HERE IS THE SOURCE OF OVERLAP
                      beatBytes = site(MemoryBusKey).beatBytes,
                      idBits = 4)
  case ExtBus => MasterPortParams(
                      base = x"6000_0000",
                      size = x"2000_0000",
                      beatBytes = site(MemoryBusKey).beatBytes,
                      idBits = 4)
  case ExtIn  => SlavePortParams(beatBytes = 8, idBits = 8, sourceBits = 4)
}))
```
I wanted to set the external memory size to zero but I got the following error.

> Caused by: java.lang.IllegalArgumentException: requirement failed: Mis-aligned AddressSets are forbidden, got: (2147483648, -1)
> 	at scala.Predef$.require(Predef.scala:224)
> 	at freechips.rocketchip.diplomacy.AddressSet.<init>(Parameters.scala:114)

Finally I put the external memory at the end of the address space of scratchpads. And it seems to be working (at least compiling).

Ssavas, 2018-08-11:
I can't seem to be running my bare-metal application on a multicore configuration with scratchpads where the ExtMem parameters are as follows (I can execute on multicore with cache):

```
case ExtMem => MasterPortParams(
                      base = x"8040_0000",
                      size = x"0FC0_0000",
                      beatBytes = site(MemoryBusKey).beatBytes,
                      idBits = 4)
```

I changed the address range due to the reason described in the above post. But I am wondering why does the external memory range overlap with the scratchpad range. And how would changing the external memory range affect the linker description which has .text.init set to 0x80000000? Because I think that might be the reason I can't execute my bare-metal application.

I tried changing the sizes of the scratchpads thinking that they might be too small. However, the result is the same. I keep getting

> *** FAILED *** via dtm (code = 1337, seed 1533977310) after 96240 cycles

Any ideas?

Uzair540, 2020-10-29:
@Ssavas  can you guide me how you put the external memory at the end of the address space of scratchpads and decrease its size accordingly.I am facing same overlapping issue

Ssavas, 2021-02-09:
> @Ssavas can you guide me how you put the external memory at the end of the address space of scratchpads and decrease its size accordingly.I am facing same overlapping issue

Hi Uzair,
 
I haven't seen your question earlier. Sorry about that. Did you fix your problem?

Regards,