chipsalliance/rocket-chip #1561: Compile ROCC emulator error

hongber, 2018-08-01:
When I use " **make CONFIG=RoCCExampleConfig**" , some errors occurred .Is there any good way to solve it?

mkdir -p /home/yhb/Desktop/rocket/rocket-chip/emulator/generated-src/
cd /home/yhb/Desktop/rocket/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/yhb/Desktop/rocket/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/yhb/Desktop/rocket/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system RoCCExampleConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project 
 from /home/yhb/Desktop/rocket/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/yhb/Desktop/rocket/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running freechips.rocketchip.system.Generator /home/yhb/Desktop/rocket/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system RoCCExampleConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "freechips.rocketchip.system.RoCCExampleConfig" from "ArrayBuffer(freechips.rocketchip.system.RoCCExampleConfig)", did you misspell it?
[error] java.lang.Exception: Unable to find part "freechips.rocketchip.system.RoCCExampleConfig" from "ArrayBuffer(freechips.rocketchip.system.RoCCExampleConfig)", did you misspell it?
[error] 	at Chisel.package$throwException$.apply(compatibility.scala:387)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:37)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:32)
[error] 	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
[error] 	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:32)
[error] 	at freechips.rocketchip.system.Generator$.getConfig(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.config(GeneratorUtils.scala:89)
[error] 	at freechips.rocketchip.system.Generator$.config$lzycompute(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.config(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.params(GeneratorUtils.scala:90)
[error] 	at freechips.rocketchip.system.Generator$.params$lzycompute(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.params(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.circuit(GeneratorUtils.scala:91)
[error] 	at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
[error] 	at freechips.rocketchip.util.GeneratorApp$class.generateFirrtl(GeneratorUtils.scala:97)
[error] 	at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:88)
[error] 	at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at freechips.rocketchip.system.Generator$.main(Generator.scala:12)
[error] 	at freechips.rocketchip.system.Generator.main(Generator.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.ClassNotFoundException: freechips.rocketchip.system.RoCCExampleConfig
[error] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
[error] 	at java.lang.Class.forName0(Native Method)
[error] 	at java.lang.Class.forName(Class.java:264)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:34)
[error] 	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:32)
[error] 	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
[error] 	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
[... 26 lines cut ...]
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] 	at sbt.Run.invokeMain(Run.scala:93)
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 7 s, completed Jul 31, 2018 1:39:35 AM
make: *** [/home/yhb/Desktop/rocket/rocket-chip/emulator/generated-src/freechips.rocketchip.system.RoCCExampleConfig.fir] Error 1

seldridge, 2018-08-01:
Try `RoccExampleConfig`. It's complaining that it can't find `RoCCExampleConfig`, which doesn't exist.

hongber, 2018-08-01:
Thank you, it has been solved.

closed 2018-08-01 by hongber