chipsalliance/rocket-chip #1563: How to modify MSTATUS_XS to run custom instruction

hongber, 2018-08-02:
I modify the file "minit.c" ,and add 
 `if (supports_extension('X'))
   set_csr(mstatus, (MSTATUS_XS & (MSTATUS_XS >> 1)));` 
 to "mstatus_init()"
then
` ./build-spike-pk.sh`
but it still can't run the custom instruction
it display:
z  0000000000000000 ra 00000000000100e2 sp 000000007f7e9b20 gp 0000000000015da8
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 000000007f7e9b50 s1 0000000000000000 a0 0000000000000001 a1 000000000000007b
a2 0000000000000002 a3 0000000000000001 a4 0000000000000010 a5 000000000000007b
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 00000000000101c2 va 0000000000c5f50b insn       00c5f50b sr 8000000200046020
An illegal instruction was executed!
Where did I go wrong?

hongber, 2018-08-02:
I have sloved it .

closed 2018-08-02 by hongber

sjtu-sp, 2018-08-31:
excuse me,
    I wanna know what did you do to solve that problem.
thank you!~