chipsalliance/rocket-chip #1566: gdb debugger

sharinoz, 2018-08-06:
hi,
I am trying to work now with the GDB debugger and I am not complete sure that I understand how to work with it
I thought it would be a good way to debug the RISCV core in the linux environment and later debug it while I run rocket on FPGA 
but, it seems to me that I debug the C file. am I right? if I am right, then what is the purpose of this debugging? 
and if I am wrong, how can I see, for example, the data stored in the registers?

thanks in advance
Shahar

sharinoz, 2018-08-20:
this is my progress so far:

rotemshahar@asic2-serv3:~$ riscv64-unknown-elf-gdb riscv-isa-sim//rot13-64
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from riscv-isa-sim//rot13-64...done.
(gdb) set remotetimeout 2000
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) load
Loading section .text, size 0x74 lma 0x10010000
Loading section .data, size 0x22 lma 0x10010078
Load failed
(gdb) print wait
Cannot access memory at address 0x1001009c
(gdb) q
A debugging session is active.

        Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/rotemshahar/riscv-isa-sim/rot13-64, Remote target
Ending remote debugging.

every time i try to load, it's almost immediately fails!
how can i fix it?
thanks
Shahar

rajeshkrmeena, 2019-01-11:
Look into OpenOCD debug log file (generate by using **-d** ). I think, you will get solution.

angi10, 2019-11-05:
I have the same issue, someone can help us?

rajeshkrmeena, 2019-11-13:
Can you share Openocd log file ( To get openocd log file, run openocd with -d and save the logs into file using -l file_name )

4ad, 2019-11-13:
Ok, so we don't confuse ourselves. At the moment I am trying to run the tests in riscv/riscv-tests using spike and openocd.

Let's pick a random test, like `DebugFunctionCall`. We can execute it manually with either `gdbserver.py targets/RISC-V/spike64.py DebugFunctionCall`, or if we want more debugging information `gdbserver.py --server_cmd "openocd -d" targets/RISC-V/spike64.py DebugFunctionCall`.

An excerpt from the log says:

    --------------------------------[ Traceback ]---------------------------------
    Traceback (most recent call last):
      File "/Users/aram/riscv-tests/debug/testlib.py", line 886, in run
        self.setup()
      File "./gdbserver.py", line 453, in setup
        self.gdb.load()
      File "/Users/aram/riscv-tests/debug/testlib.py", line 625, in load
        assert "failed" not in  output
    AssertionError

And indeed, gdb fails to parse the file:

    (gdb) load
    load
    Loading section .text, size 0x97c lma 0x1212340000
    Loading section .rodata, size 0x2b lma 0x1212340980
    Load failed

(This happens for every file, not just this particular test.)

We can run all of this manually instead of relying on `gdbserver.py`. In one terminal:

    spike -p1 --isa RV64IMAFC --dm-auth --dm-progsize 0 --dm-sba 32 --dm-abstract-rti 30 -m0x1212340000:0x10000000 --rbb-port 9824 spike64_debug-64

In another:

    REMOTE_BITBANG_HOST=localhost REMOTE_BITBANG_PORT=9824 WORK_AREA=0x1212340000 openocd -d --command 'gdb_port 52643' --command 'tcl_port disabled' --command 'telnet_port disabled' -f targets/RISC-V/spike-1.cfg

And in yet another terminal we can run gdb. All commands should be ran in the `riscv-tests/debug` directory.

Spike already complains about something, not sure if that matters or not:

    warning: tohost and fromhost symbols not in ELF; can't communicate with target

The we can reproduce the problem in gdb:

    riscv64-unknown-elf-gdb
    GNU gdb (GDB) 8.3
    Copyright (C) 2019 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "--host=x86_64-apple-darwin18.7.0 --target=riscv64-unknown-elf".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.
    
    For help, type "help".
    Type "apropos word" to search for commands related to "word".
    (gdb) set remotetimeout 30
    (gdb) target extended-remote localhost:52643
    Remote debugging using localhost:52643
    warning: No executable has been specified and target does not support
    determining executable automatically.  Try using the "file" command.
    0x000000121234095e in ?? ()
    (gdb) file spike64_debug-64
    A program is being debugged already.
    Are you sure you want to change the file? (y or n) y
    Reading symbols from spike64_debug-64...
    (gdb) load
    Loading section .text, size 0x97c lma 0x1212340000
    Loading section .rodata, size 0x2b lma 0x1212340980
    Load failed
    (gdb) 
    (gdb) 

I have attached the log from gdbserver.py, which contains the log from openocd, running with full debug info.
[20191113-155428-spike64-DebugFunctionCall.log](https://github.com/chipsalliance/rocket-chip/files/3841804/20191113-155428-spike64-DebugFunctionCall.log)

4ad, 2019-11-13:
Ok, the previous tests were all done on macOS, and all the tools (with the exception of spike) were built manually.

I redid the same tests on FreeBSD, making sure to build the exact same version of the tools, and this seems to work fine.

So some part of the toolchain seems to be broken when running on macOS.

rajeshkrmeena, 2019-11-14:
Look into the following link :

https://github.com/riscv/riscv-isa-sim/blob/master/README.md

Run Openocd with following command and share openocd.log file (when you
look into the log file you will file the reason of error)

 openocd -d -l openocd.log -f spike.cfg

4ad, 2019-11-19:
I have isolated the failure to the `--dm-progsize 0` argument passed to spike, and have filed riscv/riscv-tests#217.

closed 2020-02-09 by mwachs5