chipsalliance/rocket-chip #1567: A question about front_port.

Xuwmzzz, 2018-08-07:
Dear all,

I know front_port can read/write DTIM and ITIM.
Also it can read D-Cache.
but can it write D-Cache?

Thank you!

aswaterman, 2018-08-07:
The front port can communicate coherently with the D$, but it doesn't access the D$ directly.  Instead, it goes through the cache-coherence subsystem, which will send invalidation messages to the D$ if necessary.

Xuwmzzz, 2018-08-07:
Yes, i know the front port don't access the D$ directly.
It can use TileLink(TL-UH) to access subsystem to read the data in D$(send invalidation messages to the D$ and make the data wirteback). 
My question is that can the front port write new data to D$ by accessing subsystem?
Can the front port use 0 opcode(TileLink)  to write data when i use D$? 
Or can the front port only use 4 opcode(TileLink)  to read data when i use D$?
Thank you!

aswaterman, 2018-08-07:
Yes, you can issue writes over the front port, and they will be coherent with the D$.

Xuwmzzz, 2018-08-07:
Ok, i see. 
Thank you very much!

hcook, 2018-08-09:
Yes, in order to read the value that was written to backing memory, the cache must read from backing memory.

closed 2020-02-09 by mwachs5