chipsalliance/rocket-chip #1572: Does rocket support the debug function of access memory and quick access?

Xuwmzzz, 2018-08-10:
Dear all,

First, I delete the JTAG module from  rocket. 
Then I write a debug module with TileLink to debug(simulate) rocket by VCS.
The result show that the GPRs can be read, but the access memory operation can't be support, the cmderr bits in abstractcs(0x16) are 3'h2.
The command of access memory and quick access are optional which described in the Debug spec 0.13.
So, I want to confirm whether rocket supports these functions, or is my module wrong?
Maybe the program buffer is used to implement memory access, is it?

Thank you!

Xuwmzzz, 2018-08-10:
Does rocket support the system bus access?

WonHoYoo, 2018-10-22:
One thing I know of by simulating the rocket-chip, accessing the register has a functionality to access and pretty works fine. But the other functions the Debug Specification describes are not implemented well. Especially executing small programs inside Program Buffer seems not supportive. I guess the rocket chip wants customers implement their own way to execute the program buffer.

closed 2020-02-09 by mwachs5