chipsalliance/rocket-chip #1580: Is is possible to change dcache read path width

ShuyunJia, 2018-08-15:
Hi,
I'm looking at the Rocc's memory interface with L1D cache. The data width of this interface is derived directly from `cacheDataBits`  field of the `HasL1CacheParameters`.
I understand that a traditional CPU would use the same data path width both for register access and memory access. However, for custom extension instruction that would access memory, a wider data width(of course constrained to cache line width) would help a lot.
Is it possible to manually change this setting in the rocket chip ? I've tried to change the dcache `rowBits` as well as the `cacheDataBits` to 256 bits in my config and turns out it won't work.

hcook, 2018-08-15:
- `cacheDataBits`  defines the width of the data path going out of the cache to the memory subsystem.
- `rowDataBits` defines the width of the SRAM macro used to store cached data.
- `coreDataBits` is what you are looking for. It defines the width of the data path between the (co-)processor(s) and frontside of the cache. 

You can see `coreDataBits` used to define a signal in that interface here:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/HellaCache.scala#L101
And you can see an example of that interface being used by a RoCC here:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/tile/LazyRoCC.scala#L124

Unfortunately the existing code base is going to fight your proposal. It uses `HasCoreParameters` to ensure that the core and cache are parameterized by the same value, i.e. their interfaces will agree on the width. This constraint has the side effect of forcing all users of that interface (including RoCCs) to agree and match that width.

It will not be trivial to allow these widths to differ. I believe @jieltan is doing some work to make the RoCC IOs more parameterizable but I am not sure he has considered HellaCachIO data path width. Furthermore, the microarchitecture of the cache would have to be made to work at the widest possible width (taking into account e.g. atomic instructions) and there would have to be converters from narrow width to wide ones.

ShuyunJia, 2018-08-21:
Hi hcook,
I realized that it is a better choice to let the core and the cache agree on the width. Although this would put limitation on the cached memory bandwidth of RoCCs, it is much more important to reduce the power consumption per cache access. According to my experience in implementation, cache access accounts for a large part of the cpu's total power consumption. And a wider memory interface would increase the dynamic power proportionally. 
As to the RoCCs bandwidth, I believe it is better to use the TileLink interface which read data directly from L2. The L2 could use a inclusive strategy when RoCCs want to share data with the cores. Once L2 cache is done in Rocket-chip, I believe this won't be a problem any more.

sanket2608, 2020-01-11:
Hello,
I've some doubts regarding rocc's memory interface. 
Is it possible burst transaction between coproceessor and dcache? 
i went through accumlator example in LazyRoccScala. There is one **size** signal in **io_mem_req** interface, but it doesn't appear in the port of Accumaltorexample in verilog file.
Is that **size** signal in **HellaCacheIO** for supporting mutliburst transaction?
If multi burst is supported then is there any provision for interleaving. 
please guide me.

Thanks and regards,

ShuyunJia, 2020-01-14:
Hi @sanket2608 
I don't think the burst transaction between rocc and dcache is possible. Since the rocc access dcache via dcache's cpu req interface(with a arbitor in between). The cpu's bandwidth with dcache is limited by the instruction, which is no bigger than xLen, thus the dcache is designed that way. 

If you want to burst access dmem, it is better to use the tl_out node in Rocc, since it is a tl-c node, it would snoop into dcache and get what you want from dcache.

closed 2020-02-09 by mwachs5