chipsalliance/rocket-chip #1594: how con a rocket chip be generated for FPGAs?

rgaehfgoiarewhg, 2018-08-23:
hello

i have used the chisel generator to create a rocket chip, but the top modules is called "TestHarness". additionally i found "DPI-C" stuff in the generated verilog code. this makes it look like i have created a rocket-chip for simulations only.

how can i create a rocket chip that is ready to be put on a FPGA?

thank you very much in advance!

seldridge, 2018-08-23:
Take a look at [SiFive's Freedom repository](https://github.com/sifive/freedom) which includes infrastructure for getting Rocket Chip onto either Arty of VC707 FPGAs. 

More out of date, but also useful is [Berkeley's fpga-zynq repo](https://github.com/ucb-bar/fpga-zynq).

rgaehfgoiarewhg, 2018-08-23:
so just using the stuff on this repository it is not possible to generate anything that can be used on actual hardware? are you telling me all that fancy chisel is not able to produce anything that is actually usable?

i have already checked the links but these makefiles/generators or whatever always fail due to being outdated.

i would love nothing more than getting verilog code that can be synthesized. why is that so hard?

seldridge, 2018-08-23:
> so just using the stuff on this repository it is not possible to generate anything that can be used on actual hardware? are you telling me all that fancy chisel is not able to produce anything that is actually usable?

No, that's incorrect. Chisel produces Verilog. You can do whatever you want with that Verilog.

You can generate Verilog from this repository that forms the basis for what you need for an FPGA or an ASIC by running `make verilog` in the `vsim` directory.

However, there are a number of other steps that you have to go through _for any FPGA mapping or tape-out_, like defining a periphery for your FPGA or ASIC, replacing memories with technology specific ones, etc. _Those steps_ are handled by the repositories that I mentioned above for specific FPGAs. The referenced repositories use _this repository_ as a submodule.

rgaehfgoiarewhg, 2018-08-24:
thank you very much for the response.

yes, i understand that these repositories are supposed to ease the process, but i always have problems that make me doubt the correct functionality of my result.

i created the verilog code with the command you mentioned and got among others the file: "freechips.rocketchip.system.DefaultFPGAConfig.v". i have no idea what exactly DefaultFPGAConfig means but gave me the impression that there are other configs available. i also noticed a TestHarness in the code that sounds like it does not belong on an FPGA.

let me rephrase my problem: what should i do to adapt the rocket chip for a specific board? i assume you have to work with those .scala files right? or is there an option to get the rocket chip with some AXI interfaces that i can connect to memory?

seldridge, 2018-08-24:
Someone else may be able to provide a better explanation... but, here goes:

The actual Rocket Chip SoC is in module `ExampleRocketSystem`, which the test harness instantiates. The Rocket Chip SoC, while entirely configurable, has, for it's default configuration, AXI connections and a debug connection. By defining a different configuration and/or mixing in different traits to define your system you can change the periphery. The system is then wrapped in a test harness as it's not terrifically useful on it's own. If you want to simulate this (using Verilator, VCS, ModelSim, etc.) then you want to hook this up to simulated peripherals (an AXI memory, a debug controller, etc.). Whatever is in `ExampleRocketSystem` is _the exact_ Verilog that is going to be mapped to your FPGA or ASIC.

When you want to move to an FPGA the periphery has to be matched to some FPGA fabric periphery. Your FPGA fabric may speak PCIe, so you need to do a conversion from AXI to PCIe or avoid the conversion to AXI and go directly from TileLink to PCIe. This is only an example... This is where the existing "super repositories" I mentioned come in. Things like freedom, fpga-zynq, or firesim. These have predefined configurations for specific FPGAs: Arty, ZC707, Zedboard, ZC706, Amazon AWS ec2 f1 (small and large) for specific versions of Rocket Chip. All those repositories are encapsulated and self-starting. I only have experience with fpga-zynq due to the FPGAs I have access to, but that's literally: "type `make` in directory `X`, wait 30 minutes, get a bitstream."

Alternatively, you can take the `ExampleRocketSystem` and manually interface this with whatever you want. It speaks AXI and if you have AXI peripherals you can connect them.

> what should i do to adapt the rocket chip for a specific board? 

You need to define a shell that converts `ExampleRocketSystem` to whatever your FPGA is. SiFive's Freedom repository defines shells for specific FPGAs via the [fpga-shells](https://github.com/sifive/fpga-shells) sub-repository. That would be a good place to start poking around. In short, it's the same process you would go through for mapping any design onto any FPGA.

> i assume you have to work with those .scala files right? 

This is up to you. You can follow along with how fpga-shells works or go your own way with it. You can use the Verilog directly and write a VHDL wrapper for your FPGA. Entirely up to you, your workflow, etc.

> or is there an option to get the rocket chip with some AXI interfaces that i can connect to memory?

Take a look at the `ExampleRocketSystem` module. That already has AXI.

> i have no idea what exactly DefaultFPGAConfig means...

[`DefaultFPGAConfig`](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/Configs.scala#L82) is a specific Rocket Chip System configuration with one big core and a default periphery. That pulls in the [`BaseConfig`](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/Configs.scala#L15) configuration which is what defines that AXI + debug periphery that you see in `ExampleRocketSystem`.

rgaehfgoiarewhg, 2018-08-24:
wow, thank you very much for the detailed description. i am going to look further into it over the weekend.

minmossi, 2018-09-01:
@seldridge So, if I want to synthesize this on Zedboard right now, I have no option other than messing around sources myself?

seldridge, 2018-09-01:
@minmossi: a good starting place, if you don't care about the newest rocket-chip version, is the fpga-zynq repo.

minmossi, 2018-09-01:
@seldridge I'm afraid that repo has been deprecated and also, broken(master branch). Do you know what version I should use in order to make custom designs(altering rocket core source) without any errors?

closed 2020-02-09 by mwachs5