chipsalliance/rocket-chip #1595: sbt compile issue; object proto is not member of package firrtl

edcote, 2018-08-23:
@jackkoenig @hcook @ucbjrl and others ...

I updated rocket-chip to latest and stumbled upon the following.

Technically this may be an issue with Chisel, but I can only seem to reproduce it when building via rocket-chip.

The following can be reproduced with a simple `sbt ++2.12.6 compile`.  Noting that firrtl was compiled beforehand using `cd firrtl && sbt ++2.12.6 publishLocal`.

~~~~
[error] /home/edc/gitrepos/rocket-chip/chisel3/src/main/scala/chisel3/Driver.scala:144:12: object proto is not a member of package firrtl
[error]     firrtl.proto.ToProto.writeToStreamFast(ostream, ir.NoInfo, modules, c.name)
[error]            ^
[error] /home/edc/gitrepos/rocket-chip/chisel3/src/main/scala/chisel3/Driver.scala:209:21: reassignment to val
[error]       firrtlCircuit = Some(firrtlCircuit),
[error]                     ^
[error] two errors found
[error] (chisel / Compile / compileIncremental) Compilation failed
[error] Total time: 10 s, completed Aug 23, 2018 3:35:56 PM
~~~~

In Driver.scala, there are duplicate important to firrtl namespace.  One with proto, the other without.

~~~~
import internal.firrtl._
import firrtl._
~~~~

Working on commit a34826fad28fa32a4276d1a4f5caba63426f0656 (HEAD -> master, origin/master, origin/HEAD)

edcote, 2018-08-23:
The following change _may_ be to blame.

https://github.com/freechipsproject/rocket-chip/pull/1536

The previous commit works fine for me: HEAD detached at 6e298ea0

jackkoenig, 2018-08-24:
I was not able to reproduce this issue on a fresh clone (I did clear my local ivy cache). Did you notice this from a fresh clone or were you updating an existing repository?

Basic sanity check if it was an already existing clone, did you do a `git submodule update`?

edcote, 2018-08-24:
This was from an existing repo, but was careful with wiping cache, updating submodules, etc.

I tried again this AM, no change (commands below).  Try again now on a fresh clone.

What version of sbt are you running?
~~~~
$ history
 2038  git reset --hard origin/master
 2039  git pull origin master
 2040  git submodule update
 2041  git submodule update --init --recursive
 2045  sbt clean compile
 2046  rm -rf /home/edc/.ivy2/local/
 2047  cd firrtl/
# compile and publish firtl
 2050  sbt ++2.12.6 publishLocal
 2053  ..
# compile rocketchip
 2054  sbt ++2.12.6 compile
~~~~

edcote, 2018-08-24:
Meh, it works on a fresh clone.

After the successful compile on the fresh clone, I ran the following command to show difference between the failing build and the fresh compile.

~~~~
diff -r -q rocket-chip rocketchip-devel | egrep -v target | egrepg -v git | egrep -v project
Only in rocket-chip/emulator: generated-src
Only in rocket-chip: .idea
Only in rocket-chip: lib <---- (non-fresh clone)
~~~~

The diff is `rocket-chip/lib/firrrtl.jar`.  Not sure where that jar comes from, but its dated Apr 1.  The `.gitignore` has `/lib/` included.

Sure enough, removing the ignored directory fixes the problem.

closed 2018-08-24 by edcote

jackkoenig, 2018-08-24:
Ahhhh yes, so if there is a `firrtl.jar` in the `lib` directory, it will take priority over the published version of FIRRTL. Thus chisel was trying to build with an old version. `firrtl.jar` is created when you build rocket-chip using the Makefiles in either `vsim` or `emulator`. We hope to improve the build situation soon, but for now, that explains the issue you were seeing! Thanks for the diagnosis!