chipsalliance/rocket-chip #1611: TileLink Get fails on assertion

tampler, 2018-08-31:
Hello 

I"m trying to implement a burst RD from a SRAM Region. Here's my code snippet

```scala
class  GenericRocc1(opcodes: OpcodeSet)(implicit p: Parameters) extends LazyRoCC(opcodes, nPTWPorts = 1) {
  override lazy val module = new GenericRoccImp1(this)
  override val tlNode =  TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters("MyRocc0")))))
}
...
  // TL R/W access

  val a_WR_source = "h_A".U
  val a_RD_source = "h_B".U
  val a_address = "h_80022000".U
  val a_size = 1.U
  val a_data = "h_555".U

  val put = edge.Put(a_WR_source, a_address, a_size, a_data)._2
  val get = edge.Get(a_RD_source, a_address, a_size)._2
  val memData  = tl_out.d.bits.data
 
  tl_out.d.ready  := true.B
  tl_out.a.valid  := (state === s_tl_wr || state === s_tl_rd)
  tl_out.a.bits   := (Mux(state === s_tl_wr, put, get))

```
When I run simulation, I get the following:
Assertion failed: 'A' channel Get carries invalid source ID (connected at CrossingHelper.scala:30:80)
at Monitor.scala:65 assert (source_ok, "'A' channel Get carries invalid source ID" + extra)

What's wrong with my code ? Please advice how to fix this
Thank you

hcook, 2018-09-04:
You are creating a `TLClientNode` that claims to only have a single source id (the default), and then have a circuit that tries to use two source ids. Doing so is illegal because responses using the second id would not be routed correctly. You should construct a TLClientNode with an accurate number of source ids:
```
override val tlNode =  TLClientNode(Seq(TLClientPortParameters(Seq(
  TLClientParameters(
    name = "MyRocc0",
    sourceId      = IdRange(0, 2))))))
```

tampler, 2018-09-04:
Hello Henry

Thanks for update. I overcame that problem with some experimentation and will share my experience with TL as a 3-d party developer soon.

Meanwhile, I still can't make work outstanding (inflight) **get** transactions, which seem to be possible to implement, as per the latest SiFive TL v.1.7 spec ( p. 41, Fig. 5.3)

Can I issue **multiple outstanding** ( e.g inflight ) **GET** transactions with the current version of TileLink? I say “outstanding”, when several Channel A Get requests for the next data block are generated on the following cycle without waiting for confirmation on the Channel D

Here's my code:
```scala
class GenericRocc1(opcodes: OpcodeSet)(implicit p: Parameters) extends LazyRoCC(opcodes, nPTWPorts = 1) {
override lazy val module = new GenericRoccImp1(this)
override val tlNode = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
sourceId = IdRange (0xA,0xB),
name = s"MyCore0")))))
}
…
// TL access

val addr_mask = 0.U(5.W) // 5 bit mask to allow 32byte block transfers
val wsource = 0xA.U
val rsource = 0xA.U
val waddr = (req_vpn << pgIdxBits) + (req_offset & addr_mask) // align to 32 bytes
val raddr = ((req_vpn + ptr) << pgIdxBits) + (req_offset & addr_mask) // align to 32 bytes
val a_size = 5.U
val a_data = “h_5555”.U + ptr
val memData = chD.bits.data

val put = edge.Put(wsource, waddr, a_size, a_data)._2
val get = edge.Get(rsource, raddr, a_size)._2

chA.valid := (state === s_tl_wr || state === s_tl_rd)
chA.bits := (Mux(state === s_tl_wr, put, get))
```
When I run this, I get the following:

Req on the A interface: opc = 4, source = a, addr = 80022000, size = 5, data = 0000000000000000 
C0:     104419 [0] pc=[008000107e] W[r 0=0000000080022388][0] R[r11=0000000080022388] R[r12=0000000000000000] inst=[00c5f52b] custom1.rd.rs1.rs2 (args unknown)

Req on the A interface: opc = 4, source = a, addr = 80023000, size = 5, data = 0000000000000000 
Assertion failed: 'A' channel re-used a source ID (connected at CrossingHelper.scala:30:80)
    at Monitor.scala:448 assert(!inflight(bundle.a.bits.source), "'A' channel re-used a source ID" + extra)

Any advice is highly appreciated
Thank you!

aswaterman, 2018-09-04:
Multiple A-channel accesses are definitely supported; however, they must have distinct IDs.  You can't reuse an ID until you've received the D-channel response for that ID.

tampler, 2018-09-04:
Hi Andrew
Thanks for your reply. Okay, I made some modification and got another Assertion fail, pls see below. What's wrong with my code ?

```scala
 ...
  val wsource = 0xA.U
  val rsource = 0xA.U + ptr // <==== added an outstanding Source ID update
  val waddr = (req_vpn << pgIdxBits) + (req_offset & addr_mask) // align to 32 bytes
  val raddr = ((req_vpn + ptr) << pgIdxBits) + (req_offset & addr_mask) // align to 32 bytes
...
  val put = edge.Put(wsource, waddr, a_size, a_data)._2
  val get = edge.Get(rsource, raddr, a_size)._2
 
```
Resp on the D interface: **opc = 0**, **source = a**, sink = 0, size = 5, data = 2623fac43c23fcb4  
C0:     104418 [0] pc=[008000107e] W[r 0=0000000080022388][0] R[r11=0000000080022388] R[r12=0000000000000000] inst=[00c5f52b] custom1.rd.rs1.rs2 (args unknown)

Req on the A interface: **opc = 4**, **source = a**, **addr = 80022000**, size = 5, data = 0000000000000000 
C0:     104419 [0] pc=[008000107e] W[r 0=0000000080022388][0] R[r11=0000000080022388] R[r12=0000000000000000] inst=[00c5f52b] custom1.rd.rs1.rs2 (args unknown)

Req on the A interface: **opc = 4**, **source = b**, **addr = 80023000**, size = 5, data = 0000000000000000 
Assertion failed: **'A' channel Get carries invalid source ID** (connected at CrossingHelper.scala:30:80)
    at Monitor.scala:65 assert (source_ok, "'A' channel Get carries invalid source ID" + extra)

As you see, both source IDs and block addresses are different, but another problem occurs. 
Any ideas ?

aswaterman, 2018-09-04:
You need to expand the sourceId IdRange so that all IDs you use fall in the range [start, end).  Note that the end of the interval is not inclusive.

aswaterman, 2018-09-04:
@tampler glad to help!

closed 2018-09-05 by tampler