chipsalliance/rocket-chip #1612: ICache X data propagate in IBuf

davidmlw, 2018-09-01:
**Type of issue**: bug report

**Impact**: API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Simulate the Verilog design with ICache initialized with X value.

**What is the current behavior?**
The X values in ICache will propagate in IBuf

**What is the expected behavior?**
The X values in ICache should not propagate in IBuf.
I think there is a bug in IBuf, that is, after reset and until the first imem request is back, nBufValid should remain ZERO, and the current invalid instruction from ICache which is X at the moment should be disabled to any further logic.

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2018-09-01:
Chisel-based designs do not support X inputs. Sanitize Xs first by turning them into pseudorandom values.

closed 2018-09-01 by aswaterman

davidmlw, 2018-09-03:
I think I resolved this problem.
```scala
IBuf.scal:48      nBufValid := Mux(nReady >= nBufValid, UInt(0), nBufValid - nReady)
```
nBufValid should remain 0 when it has been 0 last cycle even when nReady is a X according to this line . But the X value from nReady will propagate through ">=" operation.

I changed this line to the following and solved the problem.
```scala
      when (nBufValid == UInt(0)){
        nBufValid := UInt(0)
      }.otherwise{
        nBufValid := Mux(nReady >= nBufValid, UInt(0), nBufValid - nReady)
      }
```