chipsalliance/rocket-chip #1618: TileLink: Increase max burst size

tampler, 2018-09-06:
Hi guys

I wanna increase the max burst size for Put/Get transactions. By default, I can get only **a_size = 6**, which corresponds to 8 beats on the 64-bit bus. If I put more, I get an assertion error: 

Assertion failed: 'A' channel carries **Get type unsupported by manager** (connected at CrossingHelper.scala:30:80)
    at Monitor.scala:64 assert (edge.manager.supportsGetSafe(edge.address(bundle), bundle.size), "'A' channel carries Get type unsupported by manager"

I digged in the TileLink sources and found that **TLFragmenter** is responsible for max burst size. However, its mostly hardcoded to **256**, such as : 
```scala
ram2.node := TLFragmenter(16, 256) := xbar2.node
```

Is there any straightforward way to increase the max burst size? I wanna support 4k, 8k, 16k and 32k multipage bursts

Thank you!

tampler, 2018-09-09:
Okay, maybe I can reformulate my question. 
How do I increase the max number of bytes that can be transferred in  a single burst. By default, I can get/put only 64 bytes ( lgSize = 6 ). How do I transfer 256 and 1024 bytes in a get/put burst?

solomatnikov, 2018-09-11:
This is in fuzzer which is only used for testing:
```
grep -n "TLFragmenter(16, 256)" `find -name *.scala`
./src/main/scala/tilelink/Fuzzer.scala:258:  ram2.node := TLFragmenter(16, 256) := xbar2.node
```

If your slave device supports long burst, then you can do long burst. Otherwise, `TLFragmenter` will break them into smaller pieces.

tampler, 2018-09-11:
Hello Alex

Thanks for your help, I appreciate. I'm having a trouble following your advice and inserting the TLFragmenter Node into the graph.
Here's my code:

```scala
class  GenericRocc(opcodes: OpcodeSet)(implicit p: Parameters) extends LazyRoCC(opcodes, nPTWPorts = 1) {
  override lazy val module = new GenericRoccImp(this)
  override val tlNode =  TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
    sourceId = IdRange (source_min, source_max),
    name = s"DMACore0")))))
}

class GenericRoccImp(outer: GenericRocc)(implicit p: Parameters) extends LazyRoCCModuleImp(outer) with HasCoreParameters  {
 
TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters( 
 
  //val tmpNode  = LazyModule ( new TLIdentityNode()) 
  val tmpNode = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(name = s"Blah")))))
  val (tl_out, tmpEdge) = outer.tlNode.out(0)
  tmpNode := TLFragmenter(8,2048) := tmpEdge

```
I get a compile error: 
[error]  cannot be applied to (freechips.rocketchip.tilelink.TLEdgeOut)
[error]   tmpNode := TLFragmenter(8,2048) := tmpEdge

How can I fix this ? More specifically, **what's the right way to insert a TLFragmenter into a ROCC module** ?

tampler, 2018-11-23:
Is there anybody from SiFive ? Please help.
I need to fragment a larger data burst with TLFragmenter and do the following:

1. Ugly, but seems to be the right way:
```scala
// LazyROCC.scala
trait HasLazyRoCC extends CanHavePTW { this: BaseTile =>

  val roccs = p(BuildRoCC).map(_(p))
  val Frag  = TLFragmenter(16,256)  

  roccs.map(_.atlNode).foreach { atl => tlMasterXbar.node :=* Frag := atl }
  roccs.map(_.tlNode).foreach { tl => tlOtherMastersNode :=* Frag := tl }

  nPTWPorts += roccs.map(_.nPTWPorts).foldLeft(0)(_ + _)
  nDCachePorts += roccs.size
}
```
2. Nicer, but still doesn't work
```scala
// ROCC implementation
class GenericRoccImp(outer: GenericRocc)(implicit p: Parameters) extends LazyRoCCModuleImp(outer) with HasCoreParameters {
 
  //val (tl_out, edge) = outer.tlNode.out(0)
  val Frag  = TLFragmenter(16,256)
  val (tl_out, tmpEdge) = outer.tlNode.out(0)
  val edge  = Frag := tmpEdge

```

Both ways provide mysterious exception trace, which is clearly due to different port signatures in TLFragmenter and LazyROCCModuleImpl.
Any ideas ?

hcook, 2018-12-15:
A `ModuleImp` can contain only generic Chisel code and parameter lookups on Edges, not Diplomatic graph construction operators applied to Nodes (which are what make Edges in the first place). Your case #1 in your more recent comment is the closest to being practical, but I would instead do something inside your own specific extension of `LazyRoCC`:
```
class  GenericRocc(opcodes: OpcodeSet)(implicit p: Parameters) extends LazyRoCC(opcodes, nPTWPorts = 1) {
  override lazy val module = new GenericRoccImp(this)
  val unfragNode =  TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
    sourceId = IdRange (source_min, source_max),
    name = s"DMACore0")))))
  tlNode :=  TLFragmenter(16,256) := unfragNode
}
```

tampler, 2018-12-16:
@hcook  
Thank you for response. I found out @ SiFive forums that putting TLFragmenter breaks my data atomicity. I don't know (yet) all other side effects, but in your code you fragment bursts near the slave RAM, where ordering and atomicity is no longer an issue.

My current approach is to add a standalone AXI interface to my ROCC. Non-cached, linear and atomic. I'll manage page alignment inside my RTL or software.

If you have a nice implementation to share, pls go ahead. Otherwise I'll PR my impl a little later.

Thanks for helping us! TileLink is awesome, but you need to put more afford on simple use cases for Diplomacy and TL. Maybe we can work out together on this...

tampler, 2020-03-12:
Not actual

closed 2020-03-12 by tampler

spiridonob, 2020-08-27:
Hi Boris, 
I'm trying to perform reading from memory to outer pin and vice versa. But I have realized that I do not comprehend Tilelink enough even just for copy any data from one memory to another. Could you show me how I can realize the most basic operation where we .Get some amount of data from one memory and .Put it to another memory?
Thank you