chipsalliance/rocket-chip #1636: Some questions about the *.out files.

hyf6661669, 2018-09-21:
Hello!

I have followed README.MD to do some tests. When I use the command "$ make output/rv64ui-p-add.out" to do a individual test, I find that the content of "rv64ui-p-add.out" is confusing. The file tells that it takes about 31000 cycles for the rocket-chip to do some wired work. And the first instruction of the ELF "rv64ui-p-add" begins to execute in cycle 31268:
C0:      31268 [1] pc=[0080000000] W[r 0=0000000080000004][1] R[r 0=0000000000000000] R[r12=0000000000000003] inst=[04c0006f] j       pc + 0x4c

And the content of cycle 483, 484, 485 is different from README.MD, my result is:
C0:        483 [0] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000000] R[r 0=0000000000000003] inst=[fe0408e3] beqz    s0, pc - 16
C0:        484 [0] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000000] R[r 0=0000000000000003] inst=[fe0408e3] beqz    s0, pc - 16
C0:        485 [0] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000000] R[r 0=0000000000000003] inst=[fe0408e3] beqz    s0, pc - 16

Besides, at the end of "rv64ui-p-add.out", it shows "Completed after 54052 cycles".

Can you tell me how to comprehend the content of the *.out files? It seems the rocket-chip spends lots of time executing instructions that has nothing to do with assembly tests.

Thanks a lot!!

mwachs5, 2018-09-21:
The code at 0x800 is the debug ROM. What you’re seeing is the debug
interface being used to load your program.

hyf6661669, 2018-09-21:
@mwachs5 
Thanks for your fast reply. 
I also find that the first instruction of different test programs starts at different cycles.
For example: 
"rv64ui-p-add" starts at cycle 31268:
C0: 31268 [1] pc=[0080000000] W[r 0=0000000080000004][1] R[r 0=0000000000000000] R[r12=0000000000000003] inst=[04c0006f] j pc + 0x4c

"rv64-ui-p-addi" starts at cycle 20630:
C0:      20630 [1] pc=[0080000000] W[r 0=0000000080000004][1] R[r 0=0000000000000000] R[r12=0000000000000003] inst=[04c0006f] j       pc + 0x4c

"rv64ui-p-jalr" starts at cycle 13289:
C0:      13289 [1] pc=[0080000000] W[r 0=0000000080000004][1] R[r 0=0000000000000000] R[r12=0000000000000003] inst=[04c0006f] j       pc + 0x4c

Does it has anything to do with the size of the test programs? I find that the number of instructions of above 3 test programs is "add" > "addi" > "jalr". I gusee it takes more time for Debug Module to load larger test programs. Am I right?

mwachs5, 2018-09-21:
Yes, it takes more cycles for the debug module to load larger programs as
they have to be written to memory one word at a time, and every word takes
many debug rom instructions to execute.

closed 2018-09-24 by hyf6661669