chipsalliance/rocket-chip #1655: error in emulator: ./emulator-freechips.rocketchip.system-DefaultConfig helloworld

ferdous313, 2018-10-03:
Hi,
I am new to rocketchip and am trying to run a helloworld using emulator.
I am getting the error and the system is halted.
Error:
"tohost and fromhost symbols not in ELF; can't communicate with target"

I run this command:
./emulator-freechips.rocketchip.system-DefaultConfig helloworld
.........
Debug Abstract Command Error #3 (EXCEPTION)ERROR: ../fesvr/dtm.cc:347, Should die, but allowing simulation to continue and fail.ERROR: ../fesvr/dtm.cc:346, Debug Abstract Command Error #3 (EXCEPTION)ERRORwarning: tohost and fromhost symbols not in ELF; can't communicate with target

Please suggest me how to solve it. Any suggestion will be appreciated very much!

aswaterman, 2018-10-03:
Have you run it on Spike before trying to run it on RTL simulation?

In both cases you probably need to use the proxy kernel.  i.e. `./emulator pk a.out` or `spike pk a.out`, rather than `./emulator a.out` or `spike a.out`

ferdous313, 2018-10-11:
I tried to set up the riscv-tool so it took me that long to answer your question. Sorry for that!
I am trying to run on spike , however I am not able to make spike pk run as it gives some error. I tried to run :
riscv-tools$riscv64-unknown-elf-gcc  -march=rv64gc -mabi=lp64 -o hello11 hello 
hello: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status

Please help me solve this.

gpsbabel, 2020-04-19:
@ferdous313 read your error carefully.
You probably meant "hello.S" or "hello.c" at the end instead of a bare "hello". That way GCC knowns which stage to feed id