chipsalliance/rocket-chip #1660: maeking verilog fails

luudee, 2018-10-05:
I followed the Readme file, but am still getting this error:

make -j8 CONFIG=DefaultSmallConfig 
mkdir -p /share/risc-v/rocket-chip/vsim/generated-src/
cd /share/risc-v/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /share/risc-v/rocket-chip/sbt-launch.jar ++2.12.4 "runMain freechips.rocketchip.system.Generator /share/risc-v/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultSmallConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /share/risc-v/rocket-chip/project
* * * *
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/rudi/.sbt/preloaded/edu.berkeley.cs/firrtl_2.12/1.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/rudi/.sbt/preloaded/edu/berkeley/cs/firrtl_2.12/1.2-SNAPSHOT/firrtl_2.12-1.2-SNAPSHOT.pom
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  ::          UNRESOLVED DEPENDENCIES         ::
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  :: edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT: not found
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]
[warn]  Note: Unresolved dependencies path:
[warn]          edu.berkeley.cs:firrtl_2.12:1.2-SNAPSHOT
[warn]            +- edu.berkeley.cs:coremacros_2.12:3.2-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT: not found

firrtl is installed and passes "sbt test" and "sbt assembly" 

What else does it want ?

The readme file needs to be updated for the build steps AND also for configuration.

Thanks !

jackkoenig, 2018-10-08:
The instructions as written work for me on both MacOS and Ubuntu, can you provide more about your platform? Are you perhaps running on Windows?

Can you also try running `find . -name 'firrtl.jar'` from the top of rocket-chip and let me know what it finds? Thanks!

jackkoenig, 2018-10-08:
Actually the part about chisel3 installation is probably unnecessary but *shouldn't* affect running `make` in `vsim`

sebastien-riou, 2018-10-12:
Same here and it is not only make verilog which fail, I get same problem for the C emulator.
I provide platform info and result of "find . -name 'firrtl.jar'"

	user@user-VirtualBox:/media/user/rocket-chip/rocket-chip/vsim$ make verilog
	make -C /media/user/rocket-chip/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /media/user/rocket-chip/rocket-chip/sbt-launch.jar ++2.12.4" root_dir=/media/user/rocket-chip/rocket-chip/firrtl build-scala
	make[1]: Entering directory '/media/user/rocket-chip/rocket-chip/firrtl'
	java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /media/user/rocket-chip/rocket-chip/sbt-launch.jar ++2.12.4 "assembly"
	OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
	[info] Loading settings from plugins.sbt ...
	[info] Loading project definition from /media/user/rocket-chip/rocket-chip/firrtl/project
	[info] Updating ProjectRef(uri("file:/media/user/rocket-chip/rocket-chip/firrtl/project/"), "firrtl-build")...
	[info] downloading https://repo1.maven.org/maven2/org/codehaus/plexus/plexus-utils/3.0.22/plexus-utils-3.0.22.jar ...
	[info] 	[SUCCESSFUL ] org.codehaus.plexus#plexus-utils;3.0.22!plexus-utils.jar (167ms)
	[info] downloading https://repo1.maven.org/maven2/org/apache/commons/commons-lang3/3.4/commons-lang3-3.4.jar ...
	[info] 	[SUCCESSFUL ] org.apache.commons#commons-lang3;3.4!commons-lang3.jar (204ms)
	[info] downloading https://repo1.maven.org/maven2/org/eclipse/sisu/org.eclipse.sisu.inject/0.3.2/org.eclipse.sisu.inject-0.3.2.jar ...
	[info] 	[SUCCESSFUL ] org.eclipse.sisu#org.eclipse.sisu.inject;0.3.2!org.eclipse.sisu.inject.jar(eclipse-plugin) (204ms)
	[info] downloading https://repo1.maven.org/maven2/org/codehaus/plexus/plexus-component-annotations/1.5.5/plexus-component-annotations-1.5.5.jar ...
	[info] 	[SUCCESSFUL ] org.codehaus.plexus#plexus-component-annotations;1.5.5!plexus-component-annotations.jar (87ms)
	[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
	[warn] 	::          UNRESOLVED DEPENDENCIES         ::
	[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
	[warn] 	:: javax.enterprise#cdi-api;1.0: Resolution failed several times for dependency: javax.enterprise#cdi-api;1.0 {compile=[compile(*), master(compile)], runtime=[runtime(*)]}:: 
	[warn] 	scalasbt: unable to get resource for org.jboss.weld#weld-api-parent;1.0: res=http://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/org.jboss.weld/weld-api-parent/1.0/jars/weld-api-parent.jar: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[warn] 	scalasbt: unable to get resource for javax.enterprise#cdi-api;1.0: res=http://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/javax.enterprise/cdi-api/1.0/ivys/ivy.xml: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[warn] 	jgit-repo: unable to get resource for javax/enterprise#cdi-api;1.0: res=http://download.eclipse.org/jgit/maven/javax/enterprise/cdi-api/1.0/cdi-api-1.0.pom: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
	[warn] 
	[warn] 	Note: Unresolved dependencies path:
	[warn] 		javax.enterprise:cdi-api:1.0
	[warn] 		  +- org.eclipse.sisu:org.eclipse.sisu.plexus:0.3.2
	[warn] 		  +- org.apache.maven:maven-plugin-api:3.3.9
	[warn] 		  +- org.pantsbuild:jarjar:1.6.5
	[warn] 		  +- com.eed3si9n:sbt-assembly:0.14.6 (scalaVersion=2.12, sbtVersion=1.0) (/media/user/rocket-chip/rocket-chip/firrtl/project/plugins.sbt#L19-20)
	[warn] 		  +- default:firrtl-build:0.1.0-SNAPSHOT (scalaVersion=2.12, sbtVersion=1.0)
	[error] sbt.librarymanagement.ResolveException: unresolved dependency: javax.enterprise#cdi-api;1.0: Resolution failed several times for dependency: javax.enterprise#cdi-api;1.0 {compile=[compile(*), master(compile)], runtime=[runtime(*)]}:: 
	[error] 	scalasbt: unable to get resource for org.jboss.weld#weld-api-parent;1.0: res=http://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/org.jboss.weld/weld-api-parent/1.0/jars/weld-api-parent.jar: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[error] 	scalasbt: unable to get resource for javax.enterprise#cdi-api;1.0: res=http://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/javax.enterprise/cdi-api/1.0/ivys/ivy.xml: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[error] 	jgit-repo: unable to get resource for javax/enterprise#cdi-api;1.0: res=http://download.eclipse.org/jgit/maven/javax/enterprise/cdi-api/1.0/cdi-api-1.0.pom: java.net.ProtocolException: Unexpected status line: Accept-Ranges: none
	[error] 
	[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
	[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
	[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
	[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
	[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
	[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
	[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
	[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
	[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[... cut ...]
	Copyright (C) 2017 Free Software Foundation, Inc.
	This is free software; see the source for copying conditions.  There is NO
	warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

	user@user-VirtualBox:/media/user/rocket-chip/rocket-chip$ java -version
	openjdk version "10.0.2" 2018-07-17
	OpenJDK Runtime Environment (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.2)
	OpenJDK 64-Bit Server VM (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.2, mixed mode)
	user@user-VirtualBox:/media/user/rocket-chip/rocket-chip$

luudee, 2018-10-14:
Sorry for the late follow up.

I'm running CentOs.

cpu11:/share/risc-v/rocket-chip >find . -name 'firrtl.jar'
./firrtl/utils/bin/firrtl.jar

Don't know what else to do ...

jackkoenig, 2018-10-15:
The Makefile is supposed to [copy `firrt.jar` to `rocket-chip/lib`](https://github.com/freechipsproject/rocket-chip/blob/b21c7879b3ea22f69cb8457109561f37c225f8ea/Makefrag#L31). 

Try deleting it and running `make` again, see if that copy is failing for some reason. You can also trying copying it yourself (although obviously you shouldn't have to do that manually).

luudee, 2018-10-15:
@jack, 
There was no "lib" directory. I created one and copied firrtl.jar into it, and got a lot further!!
Please see log below where it got stuck. And I mean stuck. Just sitting there not moving,
cpu is idle ...

Thank you for your help !

* * * 
[info] Packaging /share/risc-v/rocket-chip/target/scala-2.12/rocketchip_2.12-1.2.jar ...
[info] Done packaging.
[info] Running freechips.rocketchip.system.Generator /share/risc-v/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig
[info] [0.002] Elaborating design...
Interrupt map (2 harts 2 interrupts):
  [1, 2] => dut

[error] (Thread-17) java.io.IOException: Cannot run program "dtc": error=2, No such file or directory
[error] java.io.IOException: Cannot run program "dtc": error=2, No such file or directory
[error]         at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
[error]         at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:71)
[error]         at scala.sys.process.ProcessImpl$PipedProcesses.runAndExitValue(ProcessImpl.scala:138)
[error]         at scala.sys.process.ProcessImpl$PipedProcesses.runAndExitValue(ProcessImpl.scala:117)
[error]         at scala.sys.process.ProcessImpl$CompoundProcess.$anonfun$x$4$1(ProcessImpl.scala:96)
[error]         at scala.sys.process.ProcessImpl$Spawn$$anon$1.run(ProcessImpl.scala:23)
[error] Caused by: java.io.IOException: error=2, No such file or directory
[error]         at java.lang.UNIXProcess.forkAndExec(Native Method)
[error]         at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
[error]         at java.lang.ProcessImpl.start(ProcessImpl.java:134)
[error]         at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
[error]         at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:71)
[error]         at scala.sys.process.ProcessImpl$PipedProcesses.runAndExitValue(ProcessImpl.scala:138)
[error]         at scala.sys.process.ProcessImpl$PipedProcesses.runAndExitValue(ProcessImpl.scala:117)
[error]         at scala.sys.process.ProcessImpl$CompoundProcess.$anonfun$x$4$1(ProcessImpl.scala:96)
[error]         at scala.sys.process.ProcessImpl$Spawn$$anon$1.run(ProcessImpl.scala:23)

jackkoenig, 2018-10-15:
@luudee the Makefile was supposed to create the lib directory, not sure what went wrong!

`dtc` is the [Device Tree Compiler](https://git.kernel.org/pub/scm/utils/dtc/dtc.git). You should be able to install it with your package manager of choice. Mention of that is evidently missing from the README...

luudee, 2018-10-16:
Hi Jack, I was finally able to build it ! Many Thanks !

Do you know if there is a switch for it to generate individual verilog files?
That would enable me to perform bottom-up synthesis. Or are there any 
design compiler hooks?

Thank you !

jackkoenig, 2018-10-16:
Hi Rudolf, glad thing are working!

FIRRTL does have an option for emitting one Verilog file per module. You can add the option `--split-modules` to the FIRRTL invocation here: https://github.com/freechipsproject/rocket-chip/blob/b21c7879b3ea22f69cb8457109561f37c225f8ea/vsim/Makefrag-verilog#L17 (This can be done similarly for building in the emulator directory).

For a full list of FIRRTL options, you can invoke FIRRTL like so:
`java -cp lib/firrtl.jar firrtl.Driver --help`

jackkoenig, 2018-10-16:
Closing this, please let us know if you have other questions

closed 2018-10-16 by jackkoenig