chipsalliance/rocket-chip #1678: vsim make error

daniele1212, 2018-10-29:
Hi. After typing "cd vsim" and "make ", I got this error：

cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C
/usr/bin/ld: /home/user/synopsys/synopsys2015/vcs_mx_vJ-2014.12/amd64/lib/vcs_save_restore_new.o: relocation R_X86_64_32 against protected symbol `__start___libc_atexit' can not be used when making a PIE object
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
Makefile:95: recipe for target 'product_timestamp' failed
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory '/home/user/riscv/rocket-chip/vsim/csrc'
Make exited with status 2
Makefrag:68: recipe for target 'simv-freechips.rocketchip.system-DefaultConfig' failed
make: *** [simv-freechips.rocketchip.system-DefaultConfig] Error 2

Thanks in advance for help.

SunYongshuai, 2018-11-02:
This is the problem of vcs & gcc/g++.
When I change gcc&g++ to version 4.8, the error is gone.

1. Install gcc-4.8 & g++4.8
2. Add '-cpp g++-4.8 -cc gcc-4.8' to VCS in Makefrag
3. May be ' -LDFLAGS -Wl,-no-as-needed ' is also needed.

You can just have a try!

Mingxintang-coder, 2021-12-27:
**Makefrag in vsim fold**
VCS = vcs -full64

VCS_OPTS = -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet \
	+rad +v2k +vcs+lic+wait \
	+vc+list -CC "-I$(VCS_HOME)/include" \
	-CC "-I$(RISCV)/include" \
	-CC "-std=c++11" \
	-CC "-Wl,-rpath,$(RISCV)/lib" \
	$(RISCV)/lib/libfesvr.a \
	-sverilog \
	+incdir+$(generated_dir) \
	+define+CLOCK_PERIOD=1.0 $(sim_vsrcs) $(sim_csrcs) \
	+define+PRINTF_COND=$(TB).printf_cond \
	+define+STOP_COND=!$(TB).reset \
	+define+RANDOMIZE_MEM_INIT \
	+define+RANDOMIZE_REG_INIT \
	+define+RANDOMIZE_GARBAGE_ASSIGN \
	+define+RANDOMIZE_INVALID_ASSIGN \
	+define+RANDOMIZE_DELAY=0.1 \
	+define+MODEL=$(MODEL) \
	+libext+.v \

what can i do ?
thanks!

michael-etzkorn, 2023-01-12:
This seems to be an inactive issue about a user's environment bug. Closed.

closed 2023-01-12 by michael-etzkorn